bambulab/BambuStudio #5683: Memory leak

tonnic, 2025-01-03:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Not sure how to reproduce but it was using over 30GB of RAM.

### Actual results

[image]

### Expected results

I closed it and it released the RAM but I am not sure why there was such a huge memory leak.

### Project file & Debug log uploads

.

### Checklist of files to include

- [X] Log file
- [X] Project file

BoBoYu-1, 2025-01-06:
Thank you for your feedback. Could you please provide the reproduction steps? Or what operations did you perform before this issue occurred? Also, please provide the log file.

elbrodark, 2025-01-11:
Got the same problem, it looks like the ram fills as long as there is a camera live-feed open

bambu123, 2025-01-16:
It has been fixed, please wait for the next version.

dethdeks, 2025-03-17:
> It has been fixed, please wait for the next version.
> […](#)

nope. im on the latest version of BS and i just had to close it down as it was using over 45gigs of ram.