bambulab/BambuStudio #5687: Can't open obj file - no geometry - file works in other programs

justingolden21, 2025-01-04:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10

### Printer

P1S

### How to reproduce

Opening my obj file with adjacent mtl adjacent. Attached under project file section below.

### Actual results

[image]

[image]

[image]

### Expected results

File opens correctly

### Project file & Debug log uploads

This zip contains the obj file and mtl

[SOM Magnet.zip](https://github.com/user-attachments/files/18305874/SOM.Magnet.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

justingolden21, 2025-01-04:
After a lot of searching, I found this site: https://products.aspose.app/3d/repairing/obj

[image]

Exported from tinkercad and no other 3d software except bambu studio had any problems with it though.

Upon "repairing" it, it doesn't load the colors in bambu studio still

Haidiye00, 2025-01-06:
[image]

Hello, I downloaded the file you uploaded and found that the color-0 material is used in the obj file, but this color-0 material is not defined in MTL

justingolden21, 2025-01-06:
> Hello, I downloaded the file you uploaded and found that the color-0 material is used in the obj file, but this color-0 material is not defined in MTL

Thank you for looking into this and for the quick reply!

Do you know what could cause this? It opens fine in other programs, and I created the file in tinkercad the same as another file that opened fine in bambu. Thanks!

Haidiye00, 2025-01-06:
[image]
[image]
[image]
Hello, there is no defined color color-0 in the MTL file

justingolden21, 2025-01-06:
So is this a bug in tinkercad? Is there an easy way for me to fix this? I'm very new to 3d modeling and just trying to create something for a friend, and I've been spending days on this. Thanks again!

Haidiye00, 2025-01-08:
Hello, if you do not need multi-color, please delete the MTL file and import it. Otherwise, you will need to repair the MTL file.

justingolden21, 2025-01-28:
I need multicolor

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-03-26 by Haidiye00