bambulab/BambuStudio #5690: Rotation Coordinates not kept

contactzero, 2025-01-04:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac Sequoia 15.1.1

### Printer

Bambu A1 mini

### How to reproduce

_click printable object
_Click Rotate
_Rotate any amount of degrees

### Actual results

Rotation coordinates return 0.00.
Can reset coordinates but can't offset or compare to other parts.

### Expected results

Rotate coordinates displaced remain on the screen eg. rotate 45º, 45.00 stays in the box instead of 0.00
[Mailbox window v13.3mf.zip](https://github.com/user-attachments/files/18307244/Mailbox.window.v13.3mf.zip)

### Project file & Debug log uploads

project file

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2025-01-06:
[image]
Hello, this issue has been fixed in our internal version. Please wait for the next version

Haidiye00, 2025-01-10:
Hello, the new beta version has been released. Welcome to download and experience it~ 
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-10 by Haidiye00