bambulab/BambuStudio #5692: HDR causes BambuStudio to crash on startup

gamer, 2025-01-04:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: AMD 7800x3d 
GPU: AMD RX6950XT

### Printer

Bambu Lab P1S

### How to reproduce

1. Enable HDR in Windows display settings
2. launch BambuStudio (also crashes OrcaStudio)

### Actual results

The loading status popup will appear, then only the home row UI (home icon, prepare, preview etc) and above will appear then there is a crash without any error.

Crash log shows "ACCESS_VIOLATION".

Disabling HDR fixes the problem.

### Expected results

launch as normal

### Project file & Debug log uploads

[crash_Sat_Jan_04_18_34_11_0_done.log](https://github.com/user-attachments/files/18307878/crash_Sat_Jan_04_18_34_11_0_done.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Haidiye00, 2025-01-13:
Hello, thank you for discovering the issue. We will try to reproduce your problem and solve it~

Haidiye00, 2025-01-13:
Hello, we tried to reproduce it using an AMD graphics card with Win11, but there was no crash. Could you please upload some crash logs or dmp files for us based on this image? If there are dmp files, that would be great. Thank you~
[image]
[image]

mwarrenus, 2025-04-05:
Bambu Studio v02.00.01.50 only shows a black screen when Windows 11 HRD is enabled.  This makes Bambu Studio unusable as nothing is visible.  
Detailed PC version information (may or may not be relevant): Windows 11 Pro version 23H2, AMD Radeon RX 5700 XT GPU, driver 32.0.11027.1000.

Screenshots of the problem. Windows HDR on => Bambu Studio black screen error. Windows HDR off => Bambu Studio OK.
[image]

[image]

[image]

[image]