bambulab/BambuStudio #5696: Live View and Timelapse Not working either on the app or laptop

DhannyUB, 2025-01-05:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Sequoia 15.1.1

### Additional system information

Chip Apple M4 Max
Memory 36GB

### Printer

Bambu Lab X1C

### How to reproduce

1. Open Bambu Studio or Bambu Handy
2. Go to device
3. Select play in camera or timelapse 

### Actual results

Camera doesn't work and can't access timelapse videos from SD card

### Expected results

See live view and open my timelapse videos

### Project file & Debug log uploads

No project is failing is a software issue

### Checklist of files to include

- [ ] Log file
- [ ] Project file

c0desurfer, 2025-01-06:
Probably same as https://github.com/bambulab/BambuStudio/issues/5642.

bambu123, 2025-01-07:
Yes. You can disable firewall or wait for update.

DhannyUB, 2025-01-07:
It’s a different issue. The error is “Timeoutexception after 0:00:10.000000: liveview timeout” happens in the app and pc.

bambu123, 2025-01-07:
Timeoutexception is handy.
We are talking about Studio.
What is the error in Studio? Could you provide logs?