bambulab/BambuStudio #5697: Bambu Studio crash on startup on second monitor

Kullitschka, 2025-01-05:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: 5800X, RAM: 64GB (3600), Card: RX6700XT

### Printer

Bambu Lab P1S

### How to reproduce

If the window is moved from Bambu Studio to a second monitor and closed there, the program will no longer respond the next time it is started. I only noticed this after restarting the computer. A reinstallation does not help here.

The only option is to close the program, unplug all other monitors and restart Bambu Studio. It should then run on the main monitor again.

This seems to be a problem with Bambu Studio itself. 

### Actual results

--

### Expected results

--

### Project file & Debug log uploads

--

### Checklist of files to include

- [X] Log file
- [X] Project file

Galliyummy, 2025-01-22:
this is really annoying because my laptop monitor is broken, and it doesnt work even if the laptop monitor is completely disabled.

BoBoYu-1, 2025-01-23:
Thank you for your feedback. Could you please provide us with the log and DMP files for our analysis?

Phil-Barker, 2025-03-18:
Same issue here

I moved bambu studio to my 2nd screen so I could monitor a print whilst carrying on working.
After closing the app it refuses to open now on the 2nd screen, it just hangs.

[debug_Tue_Mar_18_09_34_19_40312.log.zip](https://github.com/user-attachments/files/19315400/debug_Tue_Mar_18_09_34_19_40312.log.zip)

2 log files attached from attempting to open the app, though I don't think they show anything useful they just seem to die at a random point

Can also confirm that unplugging my 2nd screen and opening the app then works fine.

bambu123, 2025-04-17:
Please provide DMP log following the Freeze Problem in wiki.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

> From: "Phil ***@***.***>
> Date:  Tue, Mar 18, 2025, 17:42
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> Same issue here
> I moved bambu studio to my 2nd screen so I could monitor a print whilst carrying on working.
After closing the app it refuses to open now on the 2nd screen, it just hangs.
> debug_Tue_Mar_18_09_34_19_40312.log.zip<https://github.com/user-attachments/files/19315400/debug_Tue_Mar_18_09_34_19_40312.log.zip>
> 2 log files attached from attempting to open the app, though I don't think they show anything useful they just seem to die at a random point
> Can also confirm that unplugging my 2nd screen and opening the app then works fine.
> Same issue here
> I moved bambu studio to my 2nd screen so I could monitor a print whilst carrying on working.
After closing the app it refuses to open now on the 2nd screen, it just hangs.

xb4r7x, 2025-04-28:
I'm experiencing the same problem. This behavior is experienced with Fusion360 too.