bambulab/BambuStudio #5699: Setting filament shrinkage doesn't compensate the height

mzient, 2025-01-05:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Printer

Bambu P1S

### How to reproduce

1. Slice a model, note the layer count and height of the last layer
2. Change the filament shrinkage
3. Slice again, compare layer count and height of the last layer.

### Actual results

The layer count and the number of layers is exactly the same

### Expected results

The height of the last layer should be increased to compensate for the shrinkage.
There are two ways to achieve that:
- increase layer count
- increase layer height.

### Project file & Debug log uploads

N/A, this is a functional bug.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2025-01-24:
yes, filament shrinkage controls only the xy size. i will mark this be a feature.

QingZhangBambu, 2025-01-24:
concerned staffs will review this feature