bambulab/BambuStudio #5700: Filaments not usable to choose on printer and inable to do auto k factor calibration for own created filaments

marcel1099, 2025-01-05:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

[image]

### Printer

A1 mini newest Firmware

### How to reproduce

[image]

I wanted to shrink that filament menu since i dont use all of those filaments.
Deleting the Filament Presets such as "Generic PLA" and so on doesnt help, they dont disappear from that list in the screenshot too.

Also i tried to do a auto calibration of k value on my a1 mini and it isnt possible on my on filaments i created.
It always links those k values to the "Overture PLA" filament that comes from the slicer itself instead of my own created "Overture PLA BASIC" filament.

My own created filaments are not even chooseable on the display of my a1 mini, only in the slicer.

### Actual results

see above

### Expected results

There should be a way of deleting filaments to choose from.

K Factor calibration should be possible on own created filaments, it should ask for what filament you want the k factor do be choosable.

Maybe consider to apply the k factor from the filament preset where nozzle size and so on is listed? i feel like you can easily forget to set the right k factor since you always have to choose that in the other tab.

Consider to link it to a filament preset not a "filament" that can be loaded.

### Project file & Debug log uploads

dssdds

### Checklist of files to include

- [ ] Log file
- [ ] Project file

marcel1099, 2025-01-05:
My filament preset list:
[image]

marcel1099, 2025-01-05:
[image]

gaolegao-lx, 2025-01-07:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

marcel1099, 2025-01-12:
US250112539003

marcel1099, 2025-01-12:
> **Please upload log through Handy APP, and give us your ticket number**
>
> _**Upload log files through Bambu Handy**_ Log files from the Bambu printer can be uploaded via Bambu Handy over the network, [...]

US250112539003 is ticket number with bambu handy log

marcel1099, 2025-01-12:
My x1c and a1 mini arent able to choose my recently created filaments via their touch screen. If i choose the filament via bambu handy or studio, the chosen filament will disappear automaticly and the loaded filament will show no filament.

closed 2025-05-07 by BoBoYu-BamBu