bambulab/BambuStudio #5702: K Factor handling

marcel1099, 2025-01-05:
**Is your feature request related to a problem? Please describe.**
The position of the k factor value that can be only choosen where the filament is loaded and the fact that you cant manually set it, you can only let the system set it and to do that you have to use a manual calibration or a auto calibration. There is no way to edit that value later on.

**Describe the solution you'd like**
I want to manually set a k factor myself in the filament preset, not in the device tab under "filament"

marcel1099, 2025-01-05:
imo the value in the filament preset makes more sense, since that cant be forgotten its always chosen when your printing with the preset and even when you directly print from makerworld its automaticly applied to when you use your k factor.

marcel1099, 2025-01-05:
Ok i get the point that its very usefull if you have an ams.

But for people who dont have one, for every filament swap you have to change filament in the "device" tab and manually select your K Factor preset that way.

caizixian, 2025-01-10:
OrcaSlicer supports custom PA value in the filament settings. I think it's implemented as gcode `M900 K<k factor> L1000 M10 ; Override pressure advance value`

My current workaround is to just add this to my filament start gcode when I use BambuStudio.