bambulab/BambuStudio #5703: 无法导出Gcode文件

extreme00363, 2025-01-06:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows10

### Additional system information

CPU:AMD Ryzen 5 4500U with Radeon Graphics 2.38 GHz;ROM:16G

### Printer

Creality K1C 0.8 nozzle

### How to reproduce

1、添加Creality K1C 0.8 nozzle预设；2、右键添加任意模型；3、全默认参数下切片；4、导出Gcode

### Actual results

[image]

### Expected results

无法导出Gcode，连接机器后无法上传打印

### Project file & Debug log uploads

[Bambu方块.zip](https://github.com/user-attachments/files/18314050/Bambu.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2025-01-24:
我使用1.10.2版本用您提供的stl文件导出成功了。请您试试看新版本。如果还是不行的话，请提供3mf问题，我会找相关的工作人员进行复现查看。

QingZhangBambu, 2025-01-24:
[image]