bambulab/BambuStudio #5707: LAN Only Mode — I have to enter my printer's Access Code every time

danila-piatov, 2025-01-06:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.4.1 (23E224)

### Printer

Bambu Lab A1 mini

### How to reproduce

1. Enable LAN Only Mode in the pinter
2. Add the printer by IP and Access Code into Bambu Studio
3. Restart the Bambu Studio
4. Try to reconnect to the printer again

### Actual results

The printer will ask for the Access Code again.

There are a lot of discussions about this in the Internet, e.g. https://forum.bambulab.com/t/access-code-always-needs-to-be-re-entered/82201. This seems to be a regression of some sort.

### Expected results

The printer will remember the Access Code as it remembers the IP and other information about my printer

### Project file & Debug log uploads

No project file, I was just printing a file from the printer itself.

Attaching today's logs: [logs.zip](https://github.com/user-attachments/files/18324263/logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Lukrative525, 2025-01-11:
For anyone on Windows, here is a relatively simple workaround for when Bambu Studio constantly asks for your access code: download and install AutoHotkey so that you can just enter a simple keyword and have the code entered automatically. Not a perfect solution, but makes life a little easier. @danila-piatov I'd be surprised if similar tools didn't exist for macOS.

I made a .bat file with the following commands that I run to start Bambu Studio:

`
start "C:\Program Files\AutoHotkey\v2\AutoHotkey64.exe" "C:\...(path to AutoHotkey script)...\enter_access_code.ahk"
`
`
"C:\Program Files\Bambu Studio\bambu-studio.exe"
`
`
taskkill /im "AutoHotkey64.exe"
`

This ensures that the AutoHotkey script is active only while Bambu Studio is running (as long as you close Bambu Studio through its own application window).

Here's what I put in the script (enter_access_code.ahk):

`
#SingleInstance Force ; ensures there is only one instance of the script running
`
`
::code::1234abcd ; substitutes "code" with "1234abcd" (requires Enter or Space to be pressed after "code" is typed) 
`

timurbautdinov, 2025-01-17:
Ищу решение несколько недель. После того, как отрезали китайские облака, теперь приходится жить в лан онли, которые отваливается постоянно. Если проблема решится - не забудь написать сюда. Ты не одинок(