bambulab/BambuStudio #5708: Unable to start bambu-studio [nixpkg]

ndom91, 2025-01-06:
### Bambu Studio Version

01.10.01.50

### Where is the application from?

Built from source by yourself or others

### OS version

nixos

### Additional system information

```
OS: Linux 6.12 cpe:/o:nixos:nixos:24.11 24.11 (Vicuna)
CPU: (14) x64 Intel(R) Core(TM) Ultra 7 155U
Memory: 23.07 GB / 30.86 GB
Container: Yes
Shell: 5.2.37 - /run/current-system/sw/bin/bash
```

### Printer

Bambu Lab A1

### How to reproduce

Install `01.10.01.50` from `nixpkgs` (source: https://github.com/NixOS/nixpkgs/blob/nixos-unstable/pkgs/by-name/ba/bambu-studio/package.nix). I realize it isn't a 1st party packaged version. Sidenote: I also can't start any of the 1st party AppImages, they all complain about libwebkit2gtk missing. I work on a Tauri app at work so I definitely have `libwebkit2gtk-4.1` installed and available :/

### Actual results

Running `bambu-studio` results in:

```bash
▲ bambu-studio
[2025-01-06 21:44:27.857924] [0x00007f8761c21900] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1

(bambu-studio:68278): GLib-GObject-CRITICAL **: 21:44:27.904: invalid (NULL) pointer instance

(bambu-studio:68278): GLib-GObject-CRITICAL **: 21:44:27.905: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

### Expected results

Bambu starts

### Project file & Debug log uploads

~/.config/BambuStudio/log had a few files, all of which contained the following:

```
2025-01-06 21:21:57.684059[Thread 0x00007f73f1ff4900]:Instance check: Another instance not found or single-instance not set.
2025-01-06 21:21:57.686635[Thread 0x00007f73f1ff4900]:~GUI_App: enter
2025-01-06 21:21:57.686655[Thread 0x00007f73f1ff4900]:~GUI_App: destroy app_config
2025-01-06 21:21:57.686672[Thread 0x00007f73f1ff4900]:~GUI_App: exit
```

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2025-01-07:
If the Flatpak version isn’t working properly, you could try the Docker version.
https://hub.docker.com/r/linuxserver/bambustudio

I don't know much about NixOS, so you might need to rebuild it from source.

ndom91, 2025-01-07:
@MackBambu oh awesome I had no idea that existed. Although a bit convoluted to run a desktop app in the container and expose the VNC client over a browser, it does get me printing again! :joy: :pray:

corbinwunderlich, 2025-03-23:
I'm sorry I am late, but I had this issue. Bambu Studio corrupted your files in your .config (or wherever your XDG config is) and so it tried to access a null pointer. You can fix this by doing rm -rf .config/BambuStudio (although this will sign you out and delete your preferences)

exceedsystem, 2025-07-16:
@corbinwunderlich 

>You can fix this by doing rm -rf ~/.config/BambuStudio (although this will sign you out and delete your preferences)

This helpful information solved this problem for me. Thank you :)

corbinwunderlich, 2025-09-28:
> I've got the same problem, running on NixOS unstable, I've tried the removal of my ~/.config/BambuStudio/ and also my ~/.config/BambuStudio/BambuStudio.conf is 14k (not empty) like the other thread mentions.
> 
> Immediately after I enter my auth code from email .
> 
> [bambu_crash.log](https://github.com/user-attachments/files/22576186/bambu_crash.log)

To be honest I gave up on the nixpkgs version of BambuStudio. Been using nix-flatpak instead. Works way better