bambulab/BambuStudio #5709: INstall the last release and unable to boot my ubuntu 24.04

Iotnet, 2025-01-07:
### Bambu Studio Version

1.10.1

### Where is the application from?

Bambu Lab github releases

### OS version

ubuntu 24.04

### Printer

Bambu lab A1

### How to reproduce

In my Ubuntu Desktop 24.04
1. Download Appimage  Bambu studio 1.10.1
2. Apt-get install fuse
3. chmod +x appimage
4. Run appimage and download Network manager.
5. reboot ubuntu.
6. Ubuntu Stop boot al Gnome start
7. TEsted in a Lenovo and a Acer Laptop.

### Actual results

reinstall lightgdm and the ubuntu restart , it seems that all the gnome is broken. 

### Expected results

No problem running Bambu Lab Studio with my Ubuntu system.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [X] Log file
- [ ] Project file

grrrrr, 2025-01-10:
I have experienced this too.   

Fresh install of Ubuntu 24.04.1 + updates and ` Bambu_Studio_linux_ubuntu_24.04_v01.10.01.50.AppImage `  Opting to install the Network plugin.

Rebooting Ubuntucomptuer and it hags indefinitely at

`[  OK  ] Started gdm.service - GNOME Display Manager`

grrrrr, 2025-01-24:
> I have always worked with Ubuntu 24.04

And did you test on a clean install of 24.04?

MackBambu, 2025-02-05:
> And did you test on a clean install of 24.04?

No, Bambu Studio's Linux users primarily work on Ubuntu 24.04. If it doesn't function properly, there will be a lot of error reports.

Iotnet, 2025-02-05:
I did a reinstall of the desktop and it still working as for now.

Iotnet, 2025-02-05:
At the first install , it ask me to install something about NETWORK , maybe is this part that could do the damage ?

MackBambu, 2025-03-12:
This is just a dynamic library required by BambuStudio, which is stored by default in ~/.config/BambuStudio and will not make any changes to your system.

MackBambu, 2025-03-12:
> At the first install , it ask me to install something about NETWORK , maybe is this part that could do the damage ?

This is just a dynamic library required by BambuStudio, which is stored by default in ~/.config/BambuStudio and will not make any changes to your system.

closed 2025-03-12 by MackBambu

grrrrr, 2025-03-14:
it certainly is looking like it is the problem.     

- I just downloaded the OrcaSlicer appimage.   
- Computer works fine after use and a reboot
- Allow Orca to install the **Bambu** network manager
- reboot
- broken computer

So that's now 3 out of 3 attempts to install the Bambu Network Manager that results in 3 installs of Ubuntu 24.04 not booting past

`[  OK  ] Started gdm.service - GNOME Display Manager` 

>>  And did you test on a clean install of 24.04?
> No, Bambu Studio's Linux users primarily work on Ubuntu 24.04. If it doesn't work, there will be a lot of error reports.

Great to see developers not actually willing to test reported bugs or look into them despite reports of errors.    

How many reports do Bambu require before considering it a problem?     For tracking sake,  We are at 4 reported installs of Ubuntu which are rendered unbootable after installing Bambu network manager