bambulab/BambuStudio #5713: Can not send to Micro SD Card from BambuStudio

jlsims357, 2025-01-07:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 pro

### Additional system information

Device name	DESKTOP-QJBQ3OB
Processor	11th Gen Intel(R) Core(TM) i5-1145G7 @ 2.60GHz   1.50 GHz
Installed RAM	16.0 GB (15.7 GB usable)
Device ID	DE477FE3-B4C2-4217-B30F-ED88A988F37E
Product ID	00330-80000-00000-AA647
System type	64-bit operating system, x64-based processor
Pen and touch	Touch support with 10 touch points
Edition	Windows 11 Pro
Version	24H2
Installed on	‎12/‎9/‎2024
OS build	26100.2605
Experience	Windows Feature Experience Pack 1000.26100.36.0

### Printer

Bambu A1

### How to reproduce

When I go to send a file to the micro sd card it will get about 30% done and then give me this error.  This started about 3 days ago.
[image]

### Actual results

[image]

### Expected results

The expected results would be that my file would go to the SD card like it always has.

### Project file & Debug log uploads

[Adobp Box Sliced1.zip](https://github.com/user-attachments/files/18330961/Adobp.Box.Sliced1.zip)
[crash_Tue_Jan_07_02_47_43_0_done.zip](https://github.com/user-attachments/files/18331156/crash_Tue_Jan_07_02_47_43_0_done.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

jsaccc, 2025-02-26:
Having the same issue with v 1.10.2.76 running on MacOS. Signing out of the printer, Studio, and Handy app and then signing back will **temporarily** correct the behavior. For me.

jlsims357, 2025-02-26:
jsaccc, just to let you know no one ever contacted me on this issue.  I did more research and found that I needed to reformat my SD card.  That seems to have corrected the issue.

jsaccc, 2025-02-27:
Thank you very much for sharing and I’m not surprised you weren’t contacted. I’m hearing that’s a thing with Bambu. Sent from my iPhone

closed 2025-04-21 by walterwongbbl