bambulab/BambuStudio #5714: Device -> Status: AMS slot currently in use is not being refreshed

rmilkowski, 2025-01-07:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.2

### Printer

Bambu X1C + AMS

### How to reproduce

Device -> Status ; I had live view enabled. The currently used slot from AMS doesn't always auto update (it shows an old one). If I click on Device->HMS and then back on Device->Status then it will refresh and show the correct AMS slot being in use.

I had the app in full-screen mode.

### Actual results

Old (previously used) AMS slot is being shown.

### Expected results

The  app should auto refresh AMS slot, each time it is being changed.

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

rmilkowski, 2025-01-24:
I accidentally opened a duplicate bug for this - https://github.com/bambulab/BambuStudio/issues/5848
see the attached screenshots there.

walterwongbbl, 2025-02-19:
@rmilkowski 
This is a known issue, and we will work on fixing it.

rmilkowski, 2025-04-25:
This is now fixed. Thank you.

closed 2025-04-25 by rmilkowski