bambulab/BambuStudio #5719: Correct the issue where P Series printers load X Series settings

ankonaskiff17, 2025-01-08:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home 23H2

### Printer

Bambu Lab P1S

### How to reproduce

This is a descriptive problem with all P Series printers

### Actual results

Describes a machine I DO NOT have.

### Expected results

Describe the machine I DO have

Please edit the text associated with P series printers that are configured correctly but all their settings say BBL X1C as shown in my attached screen shot.
It reflects a job unfinished to me
P-Series printers should reflect XXmm High Quality @ BBL P1S (NOT BBL X1C)
X-Series printers are the ONLY printers to should reflect XXmm High Quality @ BBL X1C

That was one of my first questions after buying my P1S and I bet I have seen doens and dozens of new PIS owners ask the exact same question.

Seems to me that fixing the text would be quite simple cut and paste script that says every instance of XIC on a P1S machine, change X1C to P1S.  
It is confusing for new owners and it opens the door to people trying to edit the settings on their own which obviously you do not want.
[image]

### Project file & Debug log uploads

Not a bug nor a model

### Checklist of files to include

- [ ] Log file
- [ ] Project file

kuuay, 2025-01-08:
Indeed, but currently there are only two models in the product line, p1 and x1, while a1 has an i3 structure. p1 and x1 have the same motion principle and appearance, and their configuration files can be used interchangeably. There is no need to create a separate configuration file for p1, just call x1.The issue you noticed is actually a naming issue, and there are no other issues involved. Why are you so obsessed with it?

BoBoYu-1, 2025-01-08:
Thank you for your feedback, but this is not a Bug, both P1S and X1C can use this process preset

ankonaskiff17, 2025-01-08:
Okay
Just be mindful that with that incorrect naming, you may be swapping out printers to customers because the customer tried to correct that documentation on their own. Correct documentation protects you.

closed 2025-01-08 by ankonaskiff17

ERisEPR, 2026-10-22:
This has cased me headaches when i first bought the P1S - i do not know why this is not updated with a seperate "P1S" or a combined "P1S/X1C"