bambulab/BambuStudio #5722: Can't add custom filament PCTG

UBaer21, 2025-01-08:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Printer

P1S AMS×3

### How to reproduce

Fillaments ↦ custom filaments ↦ create ↦ PCTG 

### Actual results

[image]

### Expected results

creating the filament

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igl-x2, 2025-01-11:
[image]

UBaer21, 2025-01-11:
@igl-x2  as you can see in my screenshot this option to select a printer is not there! Because there is no PCTG profile.
After creating a PCTG filament (via json editor)  you can create these filaments.

DanBao-Bambu, 2025-01-17:
[image]

As shown in the figure, you need to first select Generic PCTG in System Filaments and click confirm, and then create a custom PCTG. Can this solution solve your problem?

UBaer21, 2025-01-17:
As i added this manually i don't have a problem anymore. But that you need to activate "system filament" to be able to add "custom filaments"  is a bug.
Also when doing it your way it doesn't work in one step, i needed to activate "generic PCTG" you can't switch over to custom but need to close that window - that still doesn't allowed the selection but i needed to open a new project (or close and reopen BS) - Then i could select the printer/copy when adding a custom filament.
How is a custom filament "custom" if it is just a copy of an existing filament ?

UBaer21, 2025-01-17:
@DanBao-Bambu could you please tell me what would be the official process to add a "custom filament"
like [Extrudr GreenTecPro](https://www.extrudr.com/de/shop-eu/products/greentec-pro/) or[ Z-polymers Tullomer ](https://z-polymers.com/product-info) where you just don't have any profile?

DanBao-Bambu, 2025-01-17:
https://wiki.bambulab.com/en/bambu-studio/create-filament
The above website is the process of creating a custom filament.

Custom Filament is indeed a copy of an existing Filament, with only a name change. However, for this type of created Filament, it has the same effect as the system Filament, and you can change its parameters. After that, it will be a brand new Filament that you can select on the printer. This is the biggest difference compared to modifying the system Filament and saving a copy.

Indeed, as you said, you need to first select and confirm in System Filament, and then enter Custom Filament before the selected Filament can take effect. The Custom Filament function was added later, and its communication with the System Filament page is not yet complete, so you need to click OK on the System Filament page. In addition, the preset types in Custom Filament are selected in System Filament to reduce the reading burden of creating the interface. After all, displaying them all can be a burden for many people who cannot use unselected Filaments.

UBaer21, 2025-01-17:
So i can not use a new type of filament?
wouldn't it be a nice idea if custom filaments have the option to create something new?

Also when creating a filament it will create a profile for each nozzle/printer. So if i make changes to my 0.6 profile they are not in the 0.8 profile.  In theory you could just add multiple compliant printer/nozzles but if you do so you can not select that filament in AMS Device tab.

As you can see here the default system filaments on the AMS are listed as 0.8 nozzle (while i am using a 0.6 nozzle) and can't be changed.

[image]

DanBao-Bambu, 2025-01-17:
If you wish to use a new filament, please let us know. After our testing, we will introduce a new type of preset that generally has good printing effects. At this time, it will be safer for you to create a custom filament based on this preset.

For those who wish to apply the same type of Filament to multiple models, you can select multiple machine presets for nozzles of different models during creation or editing, and it will create Filaments that match the corresponding nozzles for you.

UBaer21, 2025-01-17:
So i created my filament for 0.4 and 0.6 nozzle in 2024.  Now i change some settings while using the 0.6 nozzle.
When switching to 0.4 nozzle and i am left with the original from 2024 without the changes. 

So I have GreenTEC PRO (Extrudr) Filament - and i would like to use it. If i understand you right i have to select ASA or PETG to workaround these limitations?

igl-x2, 2025-01-17:
Download the Profiles from extrudr Homepage