bambulab/BambuStudio #5727: Finish Time not showing on any prints

tomrite, 2025-01-08:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

Bambu Lab X1C

### How to reproduce

Slice and send to printer, and then the finish time below the progress bar does not show. only layer height and time left
[image]

### Actual results

the finish time is blank as in the previous screenshot

### Expected results

the finish time should fill out but it is blank

### Project file & Debug log uploads

N/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lortsie, 2025-01-26:
adding a comment to vote up

Cylindric, 2025-02-03:
I'm getting this too on 1.10.2.64.

DanBao-Bambu, 2025-02-24:
There was this issue before, you can try using different screen ratios to solve it. Please send any logs showing this situation to help track the issue.

tomrite, 2025-02-24:
> There was this issue before, you can try using different screen ratios to solve it. Please send any logs showing this situation to help track the issue.

Hi @DanBao-Bambu , 

I tried multiple screen ratios as shown in the screenshots and still no finish time is listed. 

please see logs:

[log.zip](https://github.com/user-attachments/files/18944595/log.zip)

[image]
[image]
[image]

KJ61, 2025-02-25:
Not working in 1.10.2.75

sudo-battlekafer, 2025-05-13:
same issue on 2.0.3.54

Solilord, 2025-07-01:
2.1.1.52 still no ETA .... 
27" WQHD 100% Scale

Scale Change doesn't change anything