bambulab/BambuStudio #5737: 拆分功能在处理中空模型时产生非预期的结果并产生不可用的切片产物

moe001, 2025-01-10:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

无关

### Additional system information

无关

### Printer

无关

### How to reproduce

## 1. 载入附件3mf 其内部包含两个中空立方体

## 2. 使用涂色功能的剖面视图确认中空部分完整

[image]

## 3. 右键立方体，选择`拆分->拆分到零件` (拆分到对象同样包含类似的问题，但细节略有不同)

## 4. 使用涂色功能的剖面视图确认中空部分已损坏

[image]

## 5. 进行切片并调整预览层数证实已无中空部分

### Actual results

在拆分之后正确保留中空部分，这在处理包含大量中空独立小模型的单个文件时很重要，因为我需要单独配置每个小模型的打印参数、颜色等，以及进行自定义的热床排版，现在几乎没办法实现。

### Expected results

如复现步骤所说，模型丢失了中空，似乎中空部分被视为了独立模型，图中可见进行移动之后显露出来的，原中空部分形成的实体。

[image]

### Project file & Debug log uploads

[250110.zip](https://github.com/user-attachments/files/18369161/250110.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2025-01-13:
[image]
[image]
Hello, the small cube inside was indeed preserved after splitting, but due to the same color issue, the edges were missing. If you change the color of the small cube inside, the slicing will be done

moe001, 2025-01-14:
> [image] [image] Hello, the small cube inside was indeed preserved after splitting, but due to the same color issue, the edges were missing. If you change the color of the small cube inside, the slicing will be done

不，你的描述恰恰证明了当前软件存在缺陷。正确的切片结果应该如下图般内部中空，这是没有使用拆分直接切片的结果的截图。如图才是正确的，而不是有什么可更改颜色的小正方形。在你图示的状态下，哪怕删除了`_2`和`_4`后缀的对象，切片结果也是实心正方体，内部的中空丢失了。

[image]

Haidiye00, 2025-01-15:
Hello, you can right-click to change the type of a certain CUGE to negative part.

moe001, 2025-01-15:
> Hello, you can right-click to change the type of a certain cube to negative part.

这不是根本的解决办法，我的一些实用模型会有上百个孔洞，我没理由手动修改它们，这拆分与否导致不同的切片结果，显然应该软件进行修正，而不是懒政的把issue一关了之。

Haidiye00, 2025-01-15:
Hello, can you provide a case for a complex model

moe001, 2025-01-15:
比如说我的这个拼图模型，其每个块的内部嵌有一个数字型空腔，起到了增加透明度的作用，使得打印成品对光可以看到数字解谜提示。这个模型直接切片打印时，空腔是正常的。

而我有时候需要给各个拼图块使用不同颜色的耗材，或者在加工更大的拼图时分盘打印。我就需要将其拆分，然后分别指定耗材，或者拖入不同的盘，以发送到不同的打印机加工。在拆分之后，其中的数字空腔便会变为实体，切片后也没有了空腔。因为拼图块数非常多（这个示例为了文件大小方便传输只是5x5，但实际上可能是25x25，需要分三四盘才能打印完成），手动设置那些数字实体为“负对象”非常繁琐，并且大批量生产时很容易出错。另外涉及到多个打印盘之间的拖拽，以及使用自动摆盘等功能，哪怕使用了“负对象”来折衷，也很容易导致“负对象”和块产生错位。

[main copy.zip](https://github.com/user-attachments/files/18424243/main.copy.zip)

moe001, 2025-01-15:
未拆分时直接切片，可见镂空完好。

[image]

---

拆分后切片则镂空消失不见。

[image]

---

自动摆盘后更是滑稽，镂空都变成了实体放的到处都是，这5x5的拼图已经产生了40个镂空组件，更大尺寸的拼图产生的镂空数量更是会随着拼图指数级上升，25x25可能会产生近千块独立数字，加上我制做25x25时缩小了每个拼图块的大小，这些数字更是只有米粒大，很容易产生疏漏。

[image]

Haidiye00, 2025-01-15:
Hello, regarding your question, we will first discuss it internally~

moe001, 2025-04-11:
对于这个问题，有什么更好的解决方案了吗