bambulab/BambuStudio #5740: When opening an STL file from Finder, if there is already something on the plate, in the locks up the app.

Ir8Coder, 2025-01-10:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.1 (24B83)

### Printer

Bambu Lab P1S

### How to reproduce

1. Open something in Bambu Lab and have it on the build plate in the Prepare tab
2. Open macOS Finder and open a STL file

### Actual results

When opening a STL file from macOS Finder, if there is another file already on the build plate, it adds the file to the build plate, but it does not accept any mouse clicks to do anything. I can spin the plate, but that is it.  None of the tabs work either. Maybe there is a modal window that did not make it to the forefront?  I am unsure.

### Expected results

When opening a STL file from macOS Finder, I would expect the app to just add the stl image to build plate and allow me to modify what I need to.

### Project file & Debug log uploads

I cannot upload the file, it is proprietary files.  I also can't save the project after the STL is opened.  I have opened the STL after closing the app, so it's not the file itself.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ir8Coder, 2025-01-10:
UPDATE:
Right clicking on the app, in the macOS Dock, and clicking the project name will unlock the app. I am still not sure what is causing this, but it just annoying, not a big blocker.

Haidiye00, 2025-01-14:
hello, You can directly drag the STL file into BambuStudio's prepare interface

Ir8Coder, 2025-01-14:
Yeah, I figured out a way around the issue.  It’s just kind of annoying that app locks up and you have to click on the app in the dock to get it to work. It has done it a few other times, doing other stuff as well.