bambulab/BambuStudio #5748: Scarf joint for inner walls not working

zkcreations, 2025-01-10:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Mac OS 15.2

### Additional system information

Mac mini M4 Pro

### Printer

A1

### How to reproduce

1. Enable Contour in the Filament settings
2. hit save
3. Under quality in the seam section uncheck "Scarf joint for inner walls"
4. Slice plate.

### Actual results

The scarf operation is performed in the inner walls even though it is unchecked. 

### Expected results

I just wanted the scarf to be around the outer wall ONLY but the inner walls are showing the scarf.

### Project file & Debug log uploads

[Scarf issue.3mf.zip](https://github.com/user-attachments/files/18380107/Scarf.issue.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2025-01-27:
it's bug, i will fix it.

zkcreations, 2025-02-05:
Hello,

Can you please give me an estimate on when this will get fixed? 

Thank you,
Zvi

QingZhangBambu, 2025-02-11:
@zkcreations i've fix it. it will be released affter internal testing. 

[image]

QingZhangBambu, 2025-02-11:
I'm not sure of the exact release date.

QingZhangBambu, 2025-05-23:
@zkcreations has released, you can download the latest version to check it. Thanks for reporting this bug.

closed 2025-05-23 by QingZhangBambu