bambulab/BambuStudio #5749: 不同的墙打印顺序会造成模型表面接缝状态不同

kuuay, 2025-01-10:
### Bambu Studio Version

1.10.1

### Where is the application from?

Bambu Lab Official website

### OS version

windows11

### Additional system information

amd 5600

### Printer

p1s combo

### How to reproduce

在一个模型中，将他的打印顺序从：内墙——外墙 改为 外墙——内墙时，其表面接缝会存在明显差异

### Actual results

如图当模型处于 内墙外墙时，k值会自动生效沿着正常的结果打印，当模型处于外墙内墙时，则接缝会被撕裂凹陷，产生明显的k值偏大状态，这时候缩小k值就可以解决问题，所以当一个模型的墙处于两种打印模数时，其接缝会有好和坏两种形态，疑似算法问题
[image]
[image]
[image]

### Expected results

在不同的打印顺序下，接缝应当是准寻k值保持相同水平，而不应该产生好和坏两种形态

### Project file & Debug log uploads

[圆柱体.zip](https://github.com/user-attachments/files/18380135/default.zip)
[debug_network_Sat_Jan_11_00_30_12.log.zip](https://github.com/user-attachments/files/18380149/debug_network_Sat_Jan_11_00_30_12.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2025-01-15:
你好，并不是算法问题。而是在travel运行之后，喷嘴内腔温腔压状态不稳定，导致刚开始缺料或是溢料。内墙打印一段之后促使状态稳定，会改善这个情况。

QingZhangBambu, 2025-01-15:
我会让我们的相关工作人员查看这个能否优化

kuuay, 2025-01-15:
你好，根据您的回复是刚开始缺料和溢料，但如图这个完整的模型中，整个接缝都是有问题的，我把这个现象重复测试了几次在不同模型里，当一个模型有内外两个接缝的时候（盒体），如果选择内墙外墙那么外墙的接缝是好的内墙是损坏的，如果选择外墙内墙的话那么外墙的接缝是损坏的而内墙是好的，且从一开始到结束都是这样，并没有说有趋于稳定的情况，整个接缝都有问题，而且这个问题在以前的studio版本中似乎没有，在这个版本里，我的a1和p1在墙选择中都遇到了这个问题，所以我更倾向于是studio的问题，感谢您的回复

QingZhangBambu, 2025-01-16:
您好，
对于第一个疑问，这个缺陷的作用范围是每一层。每层的打印分为两个部分，墙，填充。infill 到墙的都含有不同长度的travel路径（一般默认下是前一层填充到这一层墙）。如果你同时打两个距离较远的件，即使维持同样的内墙-》外墙的打印顺序，也会产生类似的效果。
对于第二点，如果你在版本之间有差异，可以提供相应的版本信息和打印照片，我会进行对比排查。

kuuay, 2025-01-16:
你好，我对比了两个版本结果表现一致，很高兴能听到解答，希望能在后续版本中优化这个问题，感谢！

closed 2025-01-16 by kuuay