bambulab/BambuStudio #5753: No buildplate shown - just white.

0yzmo, 2025-01-11:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Fedora 41

### Additional system information

Tried both 1.10.1 and 1.10.2 AppImage for Fedora. Also tried 1.10.1.50 from flathub. All the same. Tried an old AppImage for Ubuntu, 1.9.7.52. Same result.

Kernel: Linux 6.12.8-200.fc41.x86_64
Uptime: 26 mins
Packages: 2464 (rpm), 52 (flatpak)
Shell: bash 5.2.32
Display (GSM780D): 3440x1440 @ 240 Hz in 34" [External]
DE: GNOME 47.2
WM: Mutter (Wayland)
WM Theme: Adwaita
Theme: Adwaita [GTK2/3/4]
Icons: Adwaita [GTK2/3/4]
Font: Cantarell (11pt) [GTK2/3/4]
Cursor: Adwaita (24px)
Terminal: Ptyxis 47.6
Terminal Font: Source Code Pro (10pt)
CPU: AMD Ryzen 7 5700X3D (16) @ 4.15 GHz
GPU: NVIDIA GeForce RTX 4060 Ti 16GB [Discrete]
Memory: 7.18 GiB / 62.71 GiB (11%)
Swap: 0 B / 8.00 GiB (0%)
Disk (/): 189.98 GiB / 1.82 TiB (10%) - btrfs
Local IP (wlp5s0): 192.168.1.6/25
Locale: en_GB.UTF-8

OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce RTX 4060 Ti/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 565.77
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6.0 NVIDIA 565.77
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 565.77
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

### Printer

Bambu Lab P1S

### How to reproduce

Just download and run. Happens every time. Open model file or just open Bambulab studio, doesn't matter.
Everything else seems to work fine.

[image]
If I disable "Multi instance rendering" from menu, it shows a black area too.
[image]

I tried starting the AppImage from bash, and it gave the following output:
[image]

### Actual results

No buildplate shown.

### Expected results

Program should start and display an empty buildplate (or model if started by clicking a model-file).

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18386821/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

AssyMcGee, 2025-01-12:
I have the same problem with my nvidia dgpu / dual monitor set up running bazzite. I have no issues on my other devices that use integrated graphics or AMD dgpus.

nielsfechtel, 2025-01-13:
Same problem here, same setup, with Nvidia GPU and two monitors

Haidiye00, 2025-01-14:
Hello, we have upgraded the rendering and expect the next major version to be released. Please wait

Single-neuron, 2025-01-18:
Same problem here :'(

guanyun-gudujian, 2025-01-19:
> Same problem here :'(

i am trying to fix this issue, still needs some efforts

0yzmo, 2025-01-19:
Launching PopOS via quickemu as a workaround. Bambustudio works perfectly in PopOS :]

guanyun-gudujian, 2025-01-20:
> Same problem here :'(

can you help me to do some debugging stuff?

guanyun-gudujian, 2025-01-20:
> Launching PopOS via quickemu as a workaround. Bambustudio works perfectly in PopOS :]

can you help me to do some debugging stuff?

0yzmo, 2025-01-20:
> > Launching PopOS via quickemu as a workaround. Bambustudio works perfectly in PopOS :]
> 
> can you help me to do some debugging stuff?

I can try, though I'm surely no expert :]

guanyun-gudujian, 2025-01-20:
> > > Launching PopOS via quickemu as a workaround. Bambustudio works perfectly in PopOS :]
> > 
> > 
> > can you help me to do some debugging stuff?
> 
> I can try, though I'm surely no expert :]

Many thanks
Firstly, pls help to check if orca slicer have the same black/ white screen issue on your device
http://www.orcaslicer.cn/

0yzmo, 2025-01-20:
> Many thanks Firstly, pls help to check if orca slicer have the same black/ white screen issue on your device http://www.orcaslicer.cn/

Orcaslicer doesn't have that setting in the menu, but perhaps it is enabled by default? Anyways Orca shows a white/grey split in plate-area, so I guess that equals the black/white on Bambu:

[image]

Also, the splash-screen-dialog when I start Orca is all white:

[image]

Single-neuron, 2025-01-20:
> > Same problem here :'(
> 
> can you help me to do some debugging stuff?

I'd be happy to. Ask for anything you need.

guanyun-gudujian, 2025-01-20:
> > > Same problem here :'(
> > 
> > 
> > can you help me to do some debugging stuff?
> 
> I'd be happy to. Ask for anything you need. Orcaslicer has the same issue.

Thank you very much
Pls let me know your os type

vargn, 2025-01-20:
Same issue here with bambustudio-bin on Arch with i3

Single-neuron, 2025-01-20:
> > > > Same problem here :'(
> > > 
> > > 
> > > can you help me to do some debugging stuff?
> > 
> > 
> > I'd be happy to. Ask for anything you need. Orcaslicer has the same issue.
> 
> Thank you very much Pls let me know your os type Windows. Linux or macos

Here's the full output of 'inxi -e':

`System:
  Host: fedora Kernel: 6.12.9-200.fc41.x86_64 arch: x86_64 bits: 64
  Desktop: GNOME v: 47.3 Distro: Fedora Linux 41 (Workstation Edition)
Machine:
  Type: Desktop Mobo: Micro-Star model: Z490-A PRO (MS-7C75) v: 1.0
    serial: <superuser required> UEFI: American Megatrends v: 2.10
    date: 05/21/2020
CPU:
  Info: 6-core model: Intel Core i5-10600K bits: 64 type: MT MCP cache:
    L2: 1.5 MiB
  Speed (MHz): avg: 3800 min/max: 800/4800 cores: 1: 3800 2: 3800 3: 3800
    4: 3800 5: 3800 6: 3800 7: 3800 8: 3800 9: 3800 10: 3800 11: 3800 12: 3800
Graphics:
  Device-1: NVIDIA TU106 [GeForce RTX 2060 SUPER] driver: nvidia v: 565.77
  Display: wayland server: X.Org v: 24.1.4 with: Xwayland v: 24.1.4
    compositor: gnome-shell driver: gpu: nvidia,nvidia-nvswitch resolution:
    1: 4096x2304~60Hz 2: 3072x1728~60Hz
  API: OpenGL v: 4.6.0 vendor: nvidia v: 565.77 renderer: NVIDIA GeForce
    RTX 2060 SUPER/PCIe/SSE2
  API: EGL Message: EGL data requires eglinfo. Check --recommends.
  Info: Tools: api: glxinfo gpu: nvidia-settings x11: xdriinfo, xdpyinfo,
    xprop, xrandr
Audio:
  Device-1: Intel Comet Lake PCH cAVS driver: snd_hda_intel
  Device-2: NVIDIA TU106 High Definition Audio driver: snd_hda_intel
  API: ALSA v: k6.12.9-200.fc41.x86_64 status: kernel-api
  Server-1: PipeWire v: 1.2.7 status: active
Network:
  Device-1: Realtek RTL8125 2.5GbE driver: r8169
  IF: enp4s0 state: up speed: 1000 Mbps duplex: full mac: 2c:f0:5d:41:26:dc
Drives:
  Local Storage: total: 3.21 TiB used: 115.88 GiB (3.5%)
  ID-1: /dev/nvme0n1 vendor: Corsair model: Force MP600 size: 465.76 GiB
  ID-2: /dev/nvme1n1 vendor: KUIJIA model: Dahua C900 PLUS M.2 2280 NVMe
    1TB SSD size: 953.87 GiB
  ID-3: /dev/sda vendor: Seagate model: ST2000DM008-2FR102 size: 1.82 TiB
Partition:
  ID-1: / size: 952.28 GiB used: 115.5 GiB (12.1%) fs: btrfs
    dev: /dev/nvme1n1p3
  ID-2: /boot size: 973.4 MiB used: 368.6 MiB (37.9%) fs: ext4
    dev: /dev/nvme1n1p2
  ID-3: /boot/efi size: 598.8 MiB used: 19.3 MiB (3.2%) fs: vfat
    dev: /dev/nvme1n1p1
  ID-4: /home size: 952.28 GiB used: 115.5 GiB (12.1%) fs: btrfs
    dev: /dev/nvme1n1p3
Swap:
  ID-1: swap-1 type: zram size: 8 GiB used: 0 KiB (0.0%) dev: /dev/zram0
Sensors:
  System Temperatures: cpu: 30.0 C pch: 28.0 C mobo: N/A
  Fan Speeds (rpm): N/A
Info:
  Memory: total: 16 GiB available: 15.53 GiB used: 2.25 GiB (14.5%)
  Processes: 366 Uptime: 3m Shell: Bash inxi: 3.3.37
`

Single-neuron, 2025-01-20:
> > > > Same problem here :'(
> > > 
> > > 
> > > can you help me to do some debugging stuff?
> > 
> > 
> > I'd be happy to. Ask for anything you need. Orcaslicer has the same issue.
> 
> Thank you very much Pls let me know your os type Windows. Linux or macos

Here's info I thought was relevant. The app works on a Pop_OS! virtual machine.

### System
- **Kernel**: `6.12.9-200.fc41.x86_64`
- **Architecture**: `x86_64` (64-bit)
- **Desktop**: GNOME `v47.3`
- **Distro**: Fedora Linux 41 (Workstation Edition)

### CPU
- **Model**: Intel Core i5-10600K
- **Cores**: 6-core (12 threads)

### Graphics
- **Device**: NVIDIA TU106 [GeForce RTX 2060 SUPER]
- **Driver**: NVIDIA `v565.77`

guanyun-gudujian, 2025-01-23:
> Here's info I thought was relevant. The app works on a Pop_OS! virtual machine.
>
> ### System
> * **Kernel**: `6.12.9-200.fc41.x86_64`
> * **Architecture**: `x86_64` (64-bit)
> * **Desktop**: GNOME `v47.3` [...]

how many graphics card on your device?

AssyMcGee, 2025-01-23:
Distro: Bazzite 41 (FROM Fedora Kinoite)
Kernel: Linux 6.12.9-203.bazzite.fc41.x86_64
CPU: 12th Gen Intel(R) Core(TM) i9-12900K (24) @ 5.20 GHz
GPU: NVIDIA GeForce RTX 3070 Ti [Discrete]
GFX Driver: 565.77
NVML: 12.565.77
Monitor 1: 2560x1440 @ 144 Hz in 27" [External] *
Monitor 2: 2560x1440 @ 144 Hz in 27" [External]           
DE: KDE Plasma 6.2.5
KWin (Wayland)

guanyun-gudujian, 2025-01-23:
> Distro: Bazzite 41 (FROM Fedora Kinoite) Kernel: Linux 6.12.9-203.bazzite.fc41.x86_64 CPU: 12th Gen Intel(R) Core(TM) i9-12900K (24) @ 5.20 GHz GPU: NVIDIA GeForce RTX 3070 Ti [Discrete] GFX Driver: [...]

only one gpu card  right?

AssyMcGee, 2025-01-23:
> only one gpu card right?

Yes only one for me.

guanyun-gudujian, 2025-01-23:
> Yes only one for me.

there is a same issue with prusa
and be addressed by this comment: https://github.com/prusa3d/PrusaSlicer/issues/13653#issuecomment-2567132082
 could you pls have a try?

guanyun-gudujian, 2025-01-23:
> > > > Distro: Bazzite 41 (FROM Fedora Kinoite) Kernel: Linux 6.12.9-203.bazzite.fc41.x86_64 CPU: 12th Gen Intel(R) Core(TM) i9-12900K (24) @ 5.20 GHz GPU: NVIDIA GeForce RTX 3070 Ti [Discrete] GFX Driver: 565.77 NVML: 12.565.77 Monitor 1: 2560x1440 @ 144 Hz in 27" [External] * Monitor 2: 2560x1440 @ 144 Hz in 27" [External] DE: KDE Plasma 6.2.5 KWin (Wayland)
> > > 
> > > 
> > > only one gpu card right?
> > 
> > 
> > Yes only one for me.
> 
> there is a same issue with prusa and be addressed by this comment: [prusa3d/PrusaSlicer#13653 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/13653#issuecomment-2567132082) could you pls have a try?

@nielsfechtel   @Single-neuron @0yzmo 
Hi, all
pls have a try

Single-neuron, 2025-01-23:
By uninstalling my current flatpak driver
`flatpak remove nvidia-567-77`
and installing a different one
`flatpak install org.freedesktop.Platform.GL.nvidia-550-144-03`
I was able to see the build plate.
However the performance is horrible with constant stuttering as soon as any 3d model is loaded.
Also running from the command line I get a bunch of errors.

[logs.txt](https://github.com/user-attachments/files/18523518/logs.txt)

guanyun-gudujian, 2025-01-23:
> By uninstalling my current flatpak driver `flatpak remove nvidia-567-77` and installing a different one `flatpak install org.freedesktop.Platform.GL.nvidia-550-144-03` I was able to see the build plate. However the performance is horrible with constant stuttering as soon as any 3d model is loaded. Also running from the command line I get a bunch of errors.
> 
> [logs.txt](https://github.com/user-attachments/files/18523518/logs.txt)
> 
> edit: PrusaSlicer works fine
Congrats  that you can see the 3D rendering results.
could you please take a screenshot?
recommend tool: screen2gif
BTW, currently there are some bugs of instancing draw. Pls disable this feature in the preference page, then restart the bambu studio and check if the performance is better.
Thanks

guanyun-gudujian, 2025-01-24:
@Single-neuron 
or try to install libnvidia-egl-wayland1

according to your logs, there is a egl warnning said that failed to create dri2 screen,

dri here means `Direct Rendering Infrastructure`, which may cause side effect on rendering performance

AssyMcGee, 2025-02-03:
So I removed all my flatpak nvidia drivers and installed only 550 and the build plate shows up now. Thank you for all the help.

> Desky@bazzite:~$ flatpak remove nvidia
Similar installed refs found for ‘nvidia’:

   1) runtime/org.freedesktop.Platform.GL32.nvidia-570-86-15/x86_64/1.4 (system)
   2) runtime/org.freedesktop.Platform.GL.nvidia-565-77/x86_64/1.4 (system)
   3) runtime/org.freedesktop.Platform.GL32.nvidia-565-77/x86_64/1.4 (system)
   4) runtime/org.freedesktop.Platform.GL.nvidia-570-86-15/x86_64/1.4 (system)
   5) runtime/org.freedesktop.Platform.GL.nvidia-550-144-03/x86_64/1.4 (system)
   6) All of the above

Which do you want to use (0 to abort)? [0-6]: 6

        ID                                                        Branch            Op
 1. [-] org.freedesktop.Platform.GL.nvidia-550-144-03             1.4               r
 2. [-] org.freedesktop.Platform.GL.nvidia-565-77                 1.4               r
 3. [-] org.freedesktop.Platform.GL.nvidia-570-86-15              1.4               r
 4. [-] org.freedesktop.Platform.GL32.nvidia-565-77               1.4               r
 5. [-] org.freedesktop.Platform.GL32.nvidia-570-86-15            1.4               r

Uninstall complete.
Desky@bazzite:~$ flatpak install org.freedesktop.Platform.GL.nvidia-550-144-03
Looking for matches…

        ID                                                        Branch             Op             Remote             Download
 1. [✓] org.freedesktop.Platform.GL.nvidia-550-144-03             1.4                i              flathub            307.9 MB / 308.4 MB

Installation complete.

>

guanyun-gudujian, 2025-02-05:
> So I removed all my flatpak nvidia drivers and installed only 550 and the build plate shows up now. Thank you for all the help.
> 
> Desky@bazzite:~$ flatpak remove nvidia Similar installed refs found for ‘nvidia’:
> 
> 1. runtime/org.freedesktop.Platform.GL32.nvidia-570-86-15/x86_64/1.4 (system)
> 2. runtime/org.freedesktop.Platform.GL.nvidia-565-77/x86_64/1.4 (system)
> 3. runtime/org.freedesktop.Platform.GL32.nvidia-565-77/x86_64/1.4 (system)
> 4. runtime/org.freedesktop.Platform.GL.nvidia-570-86-15/x86_64/1.4 (system)
> 5. runtime/org.freedesktop.Platform.GL.nvidia-550-144-03/x86_64/1.4 (system)
> 6. All of the above
> 
> Which do you want to use (0 to abort)? [0-6]: 6
> 
> ```
>     ID                                                        Branch            Op
> ```
> 
> 1. [-] org.freedesktop.Platform.GL.nvidia-550-144-03             1.4               r
> 2. [-] org.freedesktop.Platform.GL.nvidia-565-77                 1.4               r
> 3. [-] org.freedesktop.Platform.GL.nvidia-570-86-15              1.4               r
> 4. [-] org.freedesktop.Platform.GL32.nvidia-565-77               1.4               r
> 5. [-] org.freedesktop.Platform.GL32.nvidia-570-86-15            1.4               r
> 
> Uninstall complete. Desky@bazzite:~$ flatpak install org.freedesktop.Platform.GL.nvidia-550-144-03 Looking for matches…
> 
> ```
>     ID                                                        Branch             Op             Remote             Download
> ```
> 
> 1. [✓] org.freedesktop.Platform.GL.nvidia-550-144-03             1.4                i              flathub            307.9 MB / 308.4 MB
> 
> Installation complete.
> 
> _EDIT_ So it fixed Bambu Studio but I was unable to use FreeCAD. I reinstalled FreeCAD and Bambu Studio stopped showing the build plate again. :(

How did you install & run freecad?
```
- flatpak install flathub org.freecad.FreeCAD
- flatpak run org.freecad.FreeCAD
```

AssyMcGee, 2025-02-05:
> How did you install & run freecad?
>
> ```
> - flatpak install flathub org.freecad.FreeCAD
> - flatpak run org.freecad.FreeCAD
> ```

I installed FreeCAD from the software center and ran it by clicking on the icon. FreeCAD worked fine. It was only Bambu Studio that malfunctioned.

guanyun-gudujian, 2025-02-05:
> I installed FreeCAD from the software center and ran it by clicking on the icon. FreeCAD worked fine. It was only Bambu Studio that malfunctioned.

pls help to check which nvidia dirver version does freecad use on your device

plus,  try to run bambu studio through flatpak and check if it is work:

```
- flatpak install flathub com.bambulab.BambuStudio
- flatpak run com.bambulab.BambuStudio
```

AssyMcGee, 2025-02-06:
```
~$`` flatpak run com.bambulab.BambuStudio

[2025-02-05 23:24:48.761191] [0x00007f9098ed8440] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
Gtk-Message: 23:24:48.831: Failed to load module "colorreload-gtk-module"
loading network module at /home/Desky/.var/app/com.bambulab.BambuStudio/config/BambuStudio/plugins/libbambu_networking.so
after dlopen, network_module is 0x56462e245a80

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.979: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.979: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.979: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.980: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.980: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.980: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.981: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.982: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.984: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.984: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.984: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.985: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.985: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:48.985: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (bambu-studio:2): CRITICAL **: 23:24:49.082: Cannot register URI scheme wxfs more than once

** (bambu-studio:2): CRITICAL **: 23:24:49.082: Cannot register URI scheme memory more than once

** (bambu-studio:2): CRITICAL **: 23:24:49.085: Cannot register URI scheme wxfs more than once

** (bambu-studio:2): CRITICAL **: 23:24:49.085: Cannot register URI scheme memory more than once

** (bambu-studio:2): CRITICAL **: 23:24:49.088: Cannot register URI scheme wxfs more than once

** (bambu-studio:2): CRITICAL **: 23:24:49.088: Cannot register URI scheme memory more than once

(bambu-studio:2): Gtk-CRITICAL **: 23:24:49.101: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:49.103: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:49.103: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:49.103: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:49.103: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:49.103: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:49.103: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:49.103: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:49.103: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:49.107: gtk_window_resize: assertion 'height > 0' failed

[... 237 lines cut ...]

(bambu-studio:2): Gtk-CRITICAL **: 23:24:51.274: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:51.277: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:52.194: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:54.144: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:55.947: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:55.947: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:55.969: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:55.972: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:55.973: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:56.064: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 23:24:56.064: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

```
This is what I got when for Bambu Studio. How would I go about finding out what version of the NVIDIA drivers FreeCAD is using.

guanyun-gudujian, 2025-02-06:
> How would I go about finding out what version of the NVIDIA drivers FreeCAD is using

seems that on some linux devices bambu studio can only be work under specify nvidia driver,
i'll try to find the root cause

epoiate, 2025-03-31:
Just confirming that this still happens with version 2.0.1.50 and that the fix by installing `org.freedesktop.Platform.GL.nvidia-550-144-03` works. 

I'll just add that in order to stop flatpack from installing newer versions you must also run `flatpak mask "org.freedesktop.Platform.GL.nvidia-*"`

Nitric81, 2025-04-19:
V2.0.2.57 still having same issue. I had to downgrade to nvidia 550-144-03 1.4 . Im on Fedora 42

# System Details Report
---

## Report details
- **Date generated:**                              2025-04-19 18:30:30

## Hardware Information:
- **Hardware Model:**                              ASUS ROG MAXIMUS Z690 FORMULA
- **Memory:**                                      32.0 GiB
- **Processor:**                                   13th Gen Intel® Core™ i5-13600K × 20
- **Graphics:**                                    NVIDIA GeForce RTX™ 4080
- **Disk Capacity:**                               3.0 TB

## Software Information:
- **Firmware Version:**                            2305
- **OS Name:**                                     Fedora Linux 42 (Workstation Edition)
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               48
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.14.2-300.fc42.x86_64

117EM, 2025-05-09:
> Just confirming that this still happens with version 2.0.1.50 and that the fix by installing `org.freedesktop.Platform.GL.nvidia-550-144-03` works.
> 
> I'll just add that in order to stop flatpack from installing newer versions you must also run `flatpak mask "org.freedesktop.Platform.GL.nvidia-*"`

Had the same problems as mentioned in this issue. The problem still persists in V 2.0.3.54 and is still fixed by this method.

robertjmcintyre, 2025-05-10:
Hitting this on a fresh install of Fedora 42, on latest NVIDIA drivers. Found this, that allows BambuStudio to work with latest nvidia drivers, weyland, etc:

https://github.com/SoftFever/OrcaSlicer/issues/8337#issuecomment-2773716317

celstark, 2025-05-22:
Following up on @robertjmcintyre --- here's the TL;DR for folks running the flatpak (at least user install)
```
sudo cp /usr/share/glvnd/egl_vendor.d/50_mesa.json ~/
flatpak override --user com.bambulab.BambuStudio  --env=__EGL_VENDOR_LIBRARY_FILENAMES=$HOME/50_mesa.json
```

You're grabbing the nouveau (software) GL info, getting it into your home folder where the flatpak can find it, and then setting flatpak to have an environment variable point to this when Bambu Studio is launched.

g1bwt, 2025-05-22:
Works for me. Thanks for taking the time to find out about it.

0yzmo, 2025-05-24:
Thank you @celstark , that solved the problem :)
Should I close this issue?

guanyun-gudujian, 2025-05-29:
> > How would I go about finding out what version of the NVIDIA drivers FreeCAD is using
> 
> seems that on some linux devices bambu studio can only be work under specify nvidia driver, i'll try to find the root cause

seams a Wayland related bug

guanyun-gudujian, 2025-05-29:
> seams a Wayland related bug

> > How would I go about finding out what version of the NVIDIA drivers FreeCAD is using
> 
> seems that on some linux devices bambu studio can only be work under specify nvidia driver, i'll try to find the root cause

seems a Wayland related bug

guanyun-gudujian, 2025-05-29:
> > How would I go about finding out what version of the NVIDIA drivers FreeCAD is using
> 
> seems that on some linux devices bambu studio can only be work under specify nvidia driver, i'll try to find the root cause
seems a Wayland related bug

guanyun-gudujian, 2025-05-29:
@robertjmcintyre 

could you pls follow this comment to help me do some testing works?
thanks
https://github.com/bambulab/BambuStudio/issues/6985#issuecomment-2915992768

guanyun-gudujian, 2025-05-30:
@0yzmo 
Hi
pls help to check if this package can work on your device.
thanks
https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15319178759/artifacts/3219406837

0yzmo, 2025-05-30:
Hi @guanyun-gudujian ! Tried to run it, but it won't start.
`oyzmo@oyzmo-fedora:~/Downloads/BambuStudio_ubuntu-22.04_V02.00.03.54$ ./Bambu_Studio_ubuntu-22.04_V02.00.03.54.AppImage 
/tmp/.mount_Bambu_AKX5Qw/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
`

0yzmo, 2025-05-30:
Installed webkit2gt3 (Fedora installed webkit2gt4.0), thought not having to install additional libraries was one of the points for using AppImages  :)

Results: Errors, but Bambulab started with plate-area all messed up.

oyzmo@oyzmo-fedora:~/Downloads/BambuStudio_ubuntu-22.04_V02.00.03.54$ ./Bambu_Studio_ubuntu-22.04_V02.00.03.54.AppImage 
[2025-05-30 14:59:13.286818] [0x00007f2e2c3b4700] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/oyzmo/.config/BambuStudio/plugins/libbambu_networking.so
after dlopen, network_module is 0x55ac3c927b90

(bambu-studio:7001): GLib-GObject-CRITICAL **: 14:59:15.006: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.006: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:7001): GLib-GObject-CRITICAL **: 14:59:15.007: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.007: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:7001): GLib-GObject-CRITICAL **: 14:59:15.008: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.008: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:7001): GLib-GObject-CRITICAL **: 14:59:15.009: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.009: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:7001): GLib-GObject-CRITICAL **: 14:59:15.013: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.013: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:7001): GLib-GObject-CRITICAL **: 14:59:15.014: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.014: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (bambu-studio:7001): CRITICAL **: 14:59:15.133: Cannot register URI scheme wxfs more than once

** (bambu-studio:7001): CRITICAL **: 14:59:15.133: Cannot register URI scheme memory more than once

** (bambu-studio:7001): CRITICAL **: 14:59:15.135: Cannot register URI scheme wxfs more than once

** (bambu-studio:7001): CRITICAL **: 14:59:15.135: Cannot register URI scheme memory more than once

** (bambu-studio:7001): CRITICAL **: 14:59:15.137: Cannot register URI scheme wxfs more than once

** (bambu-studio:7001): CRITICAL **: 14:59:15.137: Cannot register URI scheme memory more than once

** (bambu-studio:7001): CRITICAL **: 14:59:15.140: Cannot register URI scheme wxfs more than once

** (bambu-studio:7001): CRITICAL **: 14:59:15.140: Cannot register URI scheme memory more than once

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.158: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.164: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.164: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.164: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.164: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.167: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.167: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.168: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.168: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:15.168: gtk_window_resize: assertion 'height > 0' failed
[... 479 lines cut ...]
(bambu-studio:7001): GLib-GObject-CRITICAL **: 14:59:44.074: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:44.074: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:7001): GLib-GObject-CRITICAL **: 14:59:44.087: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:44.087: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:7001): GLib-GObject-CRITICAL **: 14:59:44.088: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:44.088: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:7001): GLib-GObject-CRITICAL **: 14:59:44.103: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:44.103: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:7001): GLib-GObject-CRITICAL **: 14:59:44.107: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:44.107: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:7001): GLib-GObject-CRITICAL **: 14:59:44.107: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:7001): Gtk-CRITICAL **: 14:59:44.107: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

[image]

TheMejmun, 2025-06-06:
The latest build `bambustudio-bin 02.01.00.59-1` on the AUR fixes this issue for me.

sregister, 2025-06-07:
Hmm I'm still seeing this on `bambustudio-bin 02.01.00.59-1` from AUR

[image]

guanyun-gudujian, 2025-06-07:
> > The latest build `bambustudio-bin 02.01.00.59-1` on the AUR fixes this issue for me.
> 
> How about this package? For ubuntu22.04 https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652091
> 
> For ubuntu 24.04 https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652384
> 
> Hi [@0yzmo](https://github.com/0yzmo) Pls also help to check, thanks

@sregister  how about this?

0yzmo, 2025-06-07:
> [@sregister](https://github.com/sregister) how about this?

Works :) Showed up like tha last one first (all scrambled), but just switched back and fourth between"Prepare" and "Preview", and everything is fine - good work! 👍

guanyun-gudujian, 2025-06-07:
@0yzmo 
Maybe only by swithing to X11 mode can finally resolve this gabbage rendering issue😅
Or add some ugly hack codes 😂
https://github.com/bambulab/BambuStudio/issues/6985#issuecomment-2940114602

sregister, 2025-06-07:
> [@sregister](https://github.com/sregister) how about this?

Yep confirmed the 24.04 appimage appears to work, had it install webkit2gtk but after that everything is fine.

guanyun-gudujian, 2025-06-07:
> > > > The latest build `bambustudio-bin 02.01.00.59-1` on the AUR fixes this issue for me.
> > > 
> > > 
> > > How about this package? For ubuntu22.04 https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652091
> > > For ubuntu 24.04 https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652384
> > > Hi [@0yzmo](https://github.com/0yzmo) Pls also help to check, thanks
> > 
> > 
> > [@sregister](https://github.com/sregister) how about this?
> 
> Yep confirmed both these appimages appears to work, had it install webkit2gtk but after that everything is fine.

Copy

Thanks for your feedback

0yzmo, 2025-06-08:
I updated Fedora, saw that newest NVidia driver also was listet (Driver Version: 575.57.08). No problems after install. No more scrambled view here. @guanyun-gudujian :]