bambulab/BambuStudio #5757: Precise Z height increases prime tower size significantly

Noisyfox, 2025-01-12:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambulab P1P

### How to reproduce

1. Add some model that have top surfaces, that the height is not a whole multiply of the layer z height
2. Paint the top with a different color
3. Enable Precise Z height
4. Note the prime tower depths increases significantly

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-01-12:
This issue was reported by an Orca user here: https://github.com/SoftFever/OrcaSlicer/issues/7873

zhimin-zeng-bambulab, 2025-01-20:
The "precise z-height" automatically adjusts the last 5 layers of each object, resulting in a very small layer difference between the two objects, causing the width of the prime tower to be larger.
You can  merge the two objects into one then slice.