bambulab/BambuStudio #5761: right  click tool bar not working

obambu3d, 2025-01-12:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

bambu x1c

### How to reproduce

i just tried right clicking on the model and it didn't work it was any 3mf file

### Actual results

nothing

### Expected results

bring up the right click toolbar

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18390665/log.zip)
[Creeper.zip](https://github.com/user-attachments/files/18390666/Creeper.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2025-01-13:
[image]
Hello, can you provide the graphics card model? Thank you

obambu3d, 2025-01-13:
[image]

Haidiye00, 2025-01-14:
Okay, thank you. We will try to find a machine to verify this issue

Haidiye00, 2025-10-15:
https://github.com/bambulab/BambuStudio/releases/tag/v02.03.00.70
Hello, please update the software version.