bambulab/BambuStudio #5768: unreadable Text

paulfindus, 2025-01-13:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

win 10

### Additional system information

core I7-7700

### Printer

Bambu P1S

### How to reproduce

Slicing the model "Mechanischer Wellensimulator mit Leuchtturm & Benchy wontonnn" with "slice all"

### Actual results

after slicing all plates the result is not readble because the german word "Druckplatte" ist loger then english "plate" and reaches into the result times for this plate. 
[image]

### Expected results

I would like to read the results

### Project file & Debug log uploads

[publish.zip](https://github.com/user-attachments/files/18394577/publish.zip)
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

paulfindus, 2025-06-23:
6 months later no reaction

[image]

paulfindus, 2025-08-04:
Great work in 8 month....
no changes!!!

[image]

[image]

Haidiye00, 2025-08-04:
[image]
https://github.com/bambulab/BambuStudio/releases
hello,our newest version is this

paulfindus, 2025-08-07:
What's new in the new version? - - - New bugs...
Too many new bugs in the new 2.2.... Beta 2, so it's better to have a reading problem...

paulfindus, 2025-08-07:
I tried already, but with new errors