bambulab/BambuStudio #5770: Interfering with microsoft edge

JordanFa13, 2025-01-13:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1-Carbon

### How to reproduce

Open bambustudio. Open microsoft edge

### Actual results

Anytime I have bambustudio open and go to open microsoft edge on my computer, it causes edge to lag as something is interfering between the two programs. The computer gets quite laggy for a bit and has sometimes crashed. When I dont have bambustudio open, microsoft edge works just fine.

### Expected results

The two programs should be able to run simultaneously and not interfere with each other.

### Project file & Debug log uploads

Not applicable to current issue

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2025-01-15:
This is a tricky problem, and we should abandon the use of edge runtime.

JordanFa13, 2025-01-15:
So what does that mean for me?

bambu123, 2025-01-16:
Edge Runtime has compatibility issues with your environment. Before we replace the Edge browser, we can only change the environment. For example, reinstalling the operating system and eliminating some incompatible software.

Witawat, 2025-05-17:
How to fix ?

bambu123, 2025-05-19:
Maybe you can switch to the chrome browser.

Witawat, 2025-06-06:
> Maybe you can switch to the chrome browser.
> […](#)
> 
>  From: ***@***.***>
>  Date:  Sat, May 17, 2025, 16:36
>  To: ***@***.***>
>  Cc: ***@***.***>, ***@***.***>
>  [image]Witawat left a comment ([bambulab/BambuStudio#5770](https://github.com/bambulab/BambuStudio/issues/5770))<[#5770 (comment)](https://github.com/bambulab/BambuStudio/issues/5770#issuecomment-2888216091)>
>  How to fix ?

How to switch to the chrome browser