bambulab/BambuStudio #5775: "Flush into this object" setting does not omit filaments that are marked as "soluble" and/or "support" material

austinwright, 2025-01-14:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sonoma 14.7

### Additional system information

Not Applicable

### Printer

Bambu Lab A1 w/ AMS Lite

### How to reproduce

# STEPS TO REPRODUCE:

### Base Case

1. Put two objects on the build plate in Bambu Studio.
    - Ensure the 2nd object has some infill to print at the same layer height as the 1st object's support interface.
2. Select 3 different colors of PLA in the first 3 slots of the filament section in the sidebar.
3. Select Bambu PVA for the 4th section.
    - (Alternatively, you can select any filament, turn on 'develop mode', and edit the filament's settings to turn on `soluble material` or `support material` settings).
4. Paint some colors and supports onto the first object.
    - Make sure to set the slot 4 PVA/soluble/support filament as the support interface filament.
6. Select the 2nd object and navigate to the 'Other' tab in the 'process' section of the sidebar. Under 'Flush Options', check the `Flush into objects' infill` option, and slice the plate.
7. In the 'color scheme' panel, select the 'filament' option to graph. Uncheck all the PLA filaments, so that you only see the lines for the support filament.

ACTUAL RESULT: The support filament is not purged into the sacrificial object's infill.

### Bug Case

8. Select the 2nd object and navigate to the 'Other' tab in the 'process' section of the sidebar. Under 'Flush Options', check the `Flush into this object` option, and slice the plate. (optionally uncheck the `Flush into objects' infill` option).
9. In the 'color scheme' panel, select the 'filament' option to graph. Uncheck all the PLA filaments, so that you only see the lines for the support filament.

ACTUAL RESULT: The support filament **is purged** purged into the sacrificial object.

EXPECTED RESULT: The support filament **is not purged**  into the sacrificial object.

See [this comment](https://github.com/bambulab/BambuStudio/issues/1207#issuecomment-2588701337) on https://github.com/bambulab/BambuStudio/issues/1207 for reference.

### Actual results

Control Case: Support filament is set to a standard Bambu PLA Matte, to show where the filament would be flushed into the sacrificial object with 'flush into objects' infill' option checked.
[image]

Base Case as per steps to reproduce above: Support filament is set to Bambu PVA. The lines printed in the support material from the control case are no longer present in the sliced preview.
[image]

Bug Case as per steps to reproduce above: Support filament is set to Bambu PVA. The lines printed in the support material re-appear (with slightly different paths) when the option is switched to 'flush into this object'.
[image]

### Expected results

This, except with the 'flush into this object' option selected instead of the 'flush into objects' infill' option.
[image]

### Project file & Debug log uploads

Example project that shows the basic issue with minimal config changes: [Flush into this object bug example.3mf.zip](https://github.com/user-attachments/files/18404999/Flush.into.this.object.bug.example.3mf.zip)

Log from example project: [debug_Mon_Jan_13_09_27_36_49552.log.zip](https://github.com/user-attachments/files/18405050/debug_Mon_Jan_13_09_27_36_49552.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

austinwright, 2025-01-15:
I feel like the issue might be within ToolOrdering.cpp, at `WipingExtrusions::mark_wiping_extrusions` lines 1201-1202 and/or 1210-1212 and/or 1221-1226.

https://github.com/bambulab/BambuStudio/blob/98bfabdd1cba20520c827b15a08ebb87cdd08aba/src/libslic3r/GCode/ToolOrdering.cpp#L1201:

Analysis of 1201-1202:
```
                        if (!is_overriddable(*fill, print.config(), *object, region))
                            continue;
```
The way I'm reading this is: "if not overridable, continue.". Looking into the `is_overriddable` function deeper we have the following conditions and returns in order:
1: if the extruder for the ExtrusionEntityCollection at the given region is using a soluble filament (not support, not checking the old vs new on tool change), then this object _is not_ overridable.
2: else if the object is marked as a flush object, then this object _is_ overridable.
3. else if the object _is not_ marked as an infill flush target (or the EEC role is not erInternalInfill), then this object _is not_ overridable.
4. else this object _is_ overridable.

So, it is possible we need to adjust the logic of `is_overriddable` so that it returns false in the case where the object in question is marked as a purge object.
Possible Solution 1: We may need to account for support filament as well as soluble filament in condition `1`, however currently if I mark the filament as soluble material instead of support material, the sliced plate still shows the support filament being printed into the sacrificial object. Maybe this has to do with checking both the old filament and the new filament after tool change? Because in both cases we cannot purge into the object. I need to check if the number of erroneous purged layers is equal to the number of interface layers minus 1 - that could point to this being the issue after all.

Analysis of 1210-1212:

```
                        if ((!is_entity_overridden(fill, object, copy) && fill->total_volume() > min_infill_volume))
                        {     // this infill will be used to wipe this extruder
                            set_extruder_override(fill, object, copy, new_extruder, num_of_copies);
```

Possible solution 2: Perhaps the issue is that if an object is not already marked as overridden, then we are erroneously overriding it with the next filament - which happens to be a soluble/support filament - without checking.

Analysis of 1221-1226:

```
                if (object->config().flush_into_objects && is_infill_first == perimeters_done)
                {
                    for (const ExtrusionEntity* ee : layerm->perimeters.entities) {
                        auto* fill = dynamic_cast<const ExtrusionEntityCollection*>(ee);
                        if (is_overriddable(*fill, print.config(), *object, region) && !is_entity_overridden(fill, object, copy) && fill->total_volume() > min_infill_volume) {
                            set_extruder_override(fill, object, copy, new_extruder, num_of_copies);
```

So here I think is actually the most promising area for a solution, since it seems like the soluble/support filament is additionally being printed to some perimiters when `flush_into_objects` is true, vs not at all when `flush_into_infill` is used instead. In the outermost condition we check if flush_into_objects is true (yes for us, this is the case we are trying to fix). Then we also check if the we need to print the perimeters now - we can also assume this is true for this bug case regardless of the print_infill_first setting, because eventually the perimeters need are sliced and either way we see that the support filament is being printed into the sacrificial object. Next we again check that the object is overridable (possible solution 1 above), and also that we haven't already overridden the object (possible solution 2 above), (and also that `fill->total_volume() > min_infill_volume`).

NothFie, 2025-01-15:
@austinwright Thanks for your detailed report. This is a BUG and I'm now fixing it according to your ideas. And I find if I navigate to "Support" option and choose filament_1(or 2/3) for "Support/raft base" under "Filament for Supports", then the result is just as you expect. You can use it for now. Thanks again for your advice.

austinwright, 2025-01-15:
@NothFie - Thank you for your time and prompt response.

Are you saying you were able to work around the issue? If so could you please elaborate on how you worked around it? It would be great to get this to work as it would save me (and hopefully many others!) a ton of purge waste. I tried a few things based on your comments but was not able to work around the issue:

- I added a selection for "Support/raft base" instead of 'default', but this did not change the actual result - The support filament is still purged into the sacrificial object.
- I also juggled around the 'slot' that the support filament is in, but this also did not change the outcome.

NothFie, 2025-01-16:
@austinwright I see, indeed I have also noticed that the temporary method I mentioned earlier is only effective for the testing model mentioned above. And I also tested and found that the cause of this problem is not the one you mentioned before. So it will still take me some time to figure out the problem. When I fix it, I will mention you at once.

austinwright, 2025-01-16:
@NothFie - I made a new project with similar requirements, and was somehow able to get this to work despite similar color changes / support interfaces on the same layers as a sacrificial object. The only major differences I noted:

1. The support filament is in slot 3 (not 1 or 4) (only a possible contributor to the problem, can't rule it out yet due to observations in demo project)
2. The supported objects and the sacrificial objects have the same 'base' color (seems like this is a clear test case variable that may help identify the root cause)

Here is a screenshot of the new project showing some sacrificial objects with matching base color, and some with a different base color. Note how the matching objects don't exhibit the problem, but the mismatched ones do:

[image]
(Note that the lower sacrificial object has a differing 'base' color than the supported object, whereas the upper one does not contain any support interface lines. There are also 3 other 'base' color sacrificial objects - none of which are erroneously being purged to - so it isn't a matter of the support material having been purged enough. I have the purge volume between material types set to 700.)

Update 1: I went back to my original project (the screenshots in the OG issue description) and switched the sacrificial objects to be the same color as the supported objects, and the issue went away. 1 doesn't seem to affect the outcome, except for the fact that in the demo project attached to the issue, the objects share the base color and still exhibit the problem. The support filament is in slot 4 in the demo project, so unsure if order matters here. Regardless, 3 seems to clearly demonstrate the crux of the problem and hopefully will aid in tracking down the root cause.

Update 2: Aha! I marked 3 sacrificial objects as the 3 different PLAs. The supported object was printing 2 of those 3 colors on the layer containing the support interface. Lo and behold, only the sacrificial objects of the 3rd, unused filament erroneously received the purged support filament. Confirmed this by drawing the unused filament somewhere into that layer, the outcome was no erroneous purge in the other objects. I then swapped out a different filament to make it unused, the outcome was that the sacrificial object matching that color now receives erroneous purge.

austinwright, 2025-06-15:
Checking back in to see if this has been resolved? I have laid out the exact steps to reproduce and the exact cause. I just don't have the CPP expertise to actually make a commit to fix this properly.

NothFie, 2025-06-16:
@austinwright I'm sorry, I tried the method you described, but the issue wasn't resolved completely. It seems to be more complex and involves many other factors. I still need some time to fully fix it.