bambulab/BambuStudio #5777: Text are unreadable for dark theme on linux

klesh, 2025-01-14:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Fedora 41 Workstation Gnome Wayland

### Printer

Bambu A1 mini

### How to reproduce

1. Turn on "Dark style" from the Gnome top right corner
2. Launch Bambu Studio with any stl/3mf file
3. See the problem on the Prepare and the Preview tabs

### Actual results

[image]

### Expected results

text to be readable

### Project file & Debug log uploads

Doesn't matter

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2025-01-15:
same https://github.com/bambulab/BambuStudio/issues/4943

prodestrian, 2025-03-02:
Using Dark Mode as well and been having these issues from the start. I upgraded to the latest version 1.10 and the new STEP import dialog isn't just hard to read, it's actually invisible now. At least I could squint on the other grey-on-grey text, but I have zero idea what any of these options are. Using Ubuntu 24.04 (Bambu Studio is installed via AppImage).

[image]

I temporarily switched to Light Mode and the text is fine for me in this dialog (although I haven't checked the others):

[image]

I don't think there's any way to force an app to use different colours.

MackBambu, 2025-03-12:
fix linux dark mode
https://github.com/MackBambu/BambuStudio/actions/runs/13803476334/artifacts/2735125845

MackBambu, 2025-03-26:
Fixed in v2.0

closed 2025-03-26 by MackBambu