bambulab/BambuStudio #5779: Password Invalid

Psteeper, 2025-01-14:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.2

### Additional system information

MacBook Pro
Apple M1 Pro

### Printer

Bambu Lab A1

### How to reproduce

[image]

### Actual results

Please input the printer access code.

This happens often on most prints.
This is a direct network connection to the printer. I do not want anything going the cloud. Too many businesses are hacked nowadays.

### Expected results

I expect the application to remember to remember the printer access code.

### Project file & Debug log uploads

This happens with almost every project file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2025-02-06:
Hi. @Psteeper 
Please use version 1.10.2; we will no longer delete access codes

Psteeper, 2025-02-11:
I upgraded Bambu Studio to 
￼

When I started Bambu Studio this morning I had to entered the access code. I did 3-4 prints this morning with no problems and the app has been open on my Mac all day. When I tried to print this afternoon (about 3 hours after this morning prints) it asked for the access code again . The problem is not fixed.

￼

MacOS Version

￼

Peter Steeper
Upper Tantallon, Nova Scotia

Psteeper, 2025-03-01:
Thank you!

The last upgraded fixed the problem with the access code. It works well and doesn’t forget the code now.

￼

Peter

<Bambu Studio Version.png>

<Bambu Studio Error.png>

<MacOS Version.png>