bambulab/BambuStudio #5780: Dtopdowns not dropping

toddstrader, 2025-01-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 20.04.6 LTS

### Additional system information

I'm running xmonad as my windows manager and I cannot get dropdown menus to drop down while the application's window is tiled.  Sometime it will flicker and I can see the options for a fraction of a second, but I cannot get it to stay displayed and I cannot select anything from it.  I did however find that I can work around the problem by making the window float (meta + click and drag).  I've been using xmonad for a while now and have not encountered this behavior before.

Also I tried both the appimage and flatpak releases and they both do the same thing.

### Printer

n/a

### How to reproduce

- Start the process
- Go to Prepare
- Try to select Plate Type
- You cannot select Plate Type
- meta + click and drag window
- Now you can select Plate Type

### Actual results

Dropdowns are not droppable

### Expected results

Dropdowns are droppable

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-01-15:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you~

toddstrader, 2025-01-16:
[image]

Obviously you can't see me actually clicking in the gif, but I promise that I am when my moue is over the dropdowns.

MackBambu, 2025-01-16:
If xmonad is closed, can Bambu Studio work properly?

toddstrader, 2025-01-19:
Xmonad is my windows manager so I can't easily close it and continue using Bambu Studio.  I did test switching to the default Ubuntu windows manager and that does not encounter the problem (either with the window maximized or not).

I did however, notice that the window has two sets of window control buttons (e.g. minimize, maximize).  One is from the Ubuntu windows manager and the other appears to be from Bambu Studio.  That seems pretty wonky and whatever is going on there could possibly be the culprit for my problems with xmonad.

[image]

MackBambu, 2025-01-22:
> I did however, notice that the window has two sets of window control buttons (e.g. minimize, maximize). One is from the Ubuntu windows manager and the other appears to be from Bambu Studio. That [...]

Bambu Studio has redrawn the title bar, but it failed to hide the system's default title bar on Linux.

MackBambu, 2025-03-27:
I temporarily compiled a version with the duplicate minimize and maximize buttons removed, and it now works correctly under XMonad.
https://github.com/MackBambu/BambuStudio/actions/runs/14097611191/artifacts/2828591432

toddstrader, 2025-03-27:
Sorry, what in here do I run to start BambuStudio?  I unpacked it but don't see any likely candidates.
```
tstrader@sark:~/Downloads/bambu/destdir$ find . | grep -i bambu
tstrader@sark:~/Downloads/bambu/destdir$ ls usr/local/bin/
cjpeg     curl-config  env.sh  fax2tiff  ffprobe   libpng16-config  openssl  ppm2tiff  qdelaunay  qhull     raw2tiff  rdjpgcom  tiff2pdf  tiff2rgba  tiffcp    tiffdither  tiffinfo    tiffset    tjbench    wrjpgcom   wxrc
c_rehash  djpeg        fax2ps  ffmpeg    jpegtran  libpng-config    pal2rgb  qconvex   qhalf      qvoronoi  rbox      tiff2bw   tiff2ps   tiffcmp    tiffcrop  tiffdump    tiffmedian  tiffsplit  vdb_print  wx-config  wxrc-3.1
tstrader@sark:~/Downloads/bambu/destdir$ 
```

MackBambu, 2025-03-28:
> Sorry, what in here do I run to start BambuStudio? I unpacked it but don't see any likely candidates.
>
> ```
> tstrader@sark:~/Downloads/bambu/destdir$ find . | grep -i bambu [...]

Sorry, I sent the wrong link.
ubuntu 20.04: https://github.com/MackBambu/BambuStudio/actions/runs/14097611191/artifacts/2828751942
ubuntu 24.04: https://github.com/MackBambu/BambuStudio/actions/runs/14097611191/artifacts/2828725002

toddstrader, 2025-04-03:
Unfortunately I get a segfault:
```
Thread 1 "bambustu_main" received signal SIGSEGV, Segmentation fault.
0x0000555559ac6278 in wxObject::Ref(wxObject const&) ()
(gdb) bt
#0  0x0000555559ac6278 in wxObject::Ref(wxObject const&) ()
#1  0x0000555556c52dc3 in wxEventFunctorFunctor<wxEventTypeTag<wxSizeEvent>, Slic3r::GUI::MainFrame::MainFrame()::{lambda(wxSizeEvent&)#1}>::operator()(wxEvtHandler*, wxEvent&) ()
#2  0x0000555559b07ec1 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
#3  0x0000555559b0823e in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
#4  0x0000555559b083a4 in wxEvtHandler::TryHereOnly(wxEvent&) ()
#5  0x0000555559b0844f in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
#6  0x0000555559b084f9 in wxEvtHandler::ProcessEvent(wxEvent&) ()
#7  0x0000555559b08e7b in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
#8  0x0000555559950232 in wxTopLevelWindowGTK::DoSetSize(int, int, int, int, int) ()
#9  0x0000555559956548 in wxWindow::DoSetClientSize(int, int) ()
#10 0x000055555995198d in wxTopLevelWindowGTK::DoSetClientSize(int, int) ()
#11 0x000055555987a585 in wxWindowBase::WXSetInitialFittingClientSize(int) ()
#12 0x0000555559951e14 in wxTopLevelWindowGTK::WXSetInitialFittingClientSize(int) ()
#13 0x0000555556c628cf in Slic3r::GUI::MainFrame::MainFrame() ()
#14 0x0000555556c18d9c in Slic3r::GUI::GUI_App::on_init_inner() ()
#15 0x0000555556c19ea7 in Slic3r::GUI::GUI_App::OnInit() ()
#16 0x0000555559aa8a47 in wxEntry(int&, wchar_t**) ()
#17 0x0000555556bd0beb in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#18 0x00005555560379bf in Slic3r::CLI::run(int, char**) ()
#19 0x0000555555f1ae69 in main ()
```

I'm on:
```
tstrader@sark:~/Downloads/bambu$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal
```