bambulab/BambuStudio #5781: OpenSSL compile faild in win11

lidegao899, 2025-01-15:
### Bambu Studio Version

main

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

[image]

### Printer

Null

### How to reproduce

bbs can not compile on my computer due to OpenSSL [error](https://github.com/openssl/openssl/issues/20805).
I fix this problem by change OpenSSL version to [3.4.0](https://github.com/openssl/openssl/releases/tag/openssl-3.4.0).

[image]

### Actual results

OpenSSL dep build faild.

### Expected results

no error.

### Project file & Debug log uploads

Null

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lidegao899, 2025-01-17:
> What's the failure log?
> […](#)

This is the error msg when I run `msbuild /m ALL_BUILD.vcxproj`.
[image]

nisparks, 2025-06-14:
Same issue. Just put in the latest and seems to be working. 

set(url_str "https://github.com/openssl/openssl/archive/refs/tags/openssl-3.5.0.tar.gz")
set(url_hash "SHA256=576740BFE93209EFDA2BE7CECB00DFEABE77FE9C5C908C1790CBE8FA50B223D0")