bambulab/BambuStudio #5782: I solved why LAN mode fails constantly, and the fix is easy if you want to make it work. You forgot to listen on port 2021

skintigh, 2025-01-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home

### Additional system information

Device name	Gram
Processor	13th Gen Intel(R) Core(TM) i7-1360P   2.20 GHz
Installed RAM	32.0 GB (31.7 GB usable)
Device ID	7F1BC85B-9496-46DD-A3F9-E86230BD9570
Product ID	00342-21069-11935-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

X1C

### How to reproduce

Send. 

Press the button that says "send." Apparently you never tried this once in your testing before shipping the software.

### Actual results

Fail. Almost every time.

### Expected results

Print.

### Project file & Debug log uploads

Here's the problem: for no apparent reason you require the printer to send a packet on port 2021 before you allow file transfers, and for no apparent reason you require this action again after the transfer before you allow the printer to print.

**Except you forgot to listen on the port.**

Fate protects fools, little children and Bambu Lab as this somehow actually works when all **network security has been disabled, something that should never be done.** (You can do this by turning off the windows defender firewall, which secretly _also_ turns off Windows Filtering Platform (WFP) and it's hidden secret rules that don't log to the firewall log. Because Microsoft.)

You application has to be listening on the port in order to receive the packet. If nobody is listening, windows "Stealth" filter in WFP decides this must be unsolicited port scan or denial of service attack or and drops the packet.

That's why turning off the firewall allows your program to work and actually print in LAN mode.

That's also why creating rules is futile. No rule will ever be checked if the packet is dropped by the stealth filter. That's a special feature of windows. It will be silently dropped, and it will not be logged as being dropped by the firewall because it's not being dropped by the firewall (where firewall rules live). It's being droppe by Windows FIltering Platform. See https://serverfault.com/questions/299762/windows-platform-filtering-legitimate-packets-blocked

wfpstate.xml:

ffffffffffffffff ........ FWPM_LAYER_INBOUND_TRANSPORT_V4_DISCARD FWPM_SUBLAYER_MPSSVC_WF FWP_UINT8 12 FWPM_CONDITION_FLAGS FWP_MATCH_FLAGS_NONE_SET FWP_UINT32 3 FWP_ACTION_CALLOUT_TERMINATING FWPM_CALLOUT_WFP_TRANSPORT_LAYER_V4_SILENT_DROP {999ba50e-b870-4b5c-84e6-33a0e61cc8d1} 691749 FWP_UINT64 13835058055349272576 FWPM_LAYER_INBOUND_TRANSPORT_V6 Inbound Transport v6 Layer FWPM_LAYER_FLAG_KERNEL FWPM_LAYER_FLAG_BUILTIN FWPM_LAYER_FLAG_CLASSIFY_MOSTLY FWPM_CONDITION_IP_PROTOCOL FWPM_FIELD_RAW_DATA FWP_UINT8 FWPM_CONDITION_IP_LOCAL_ADDRESS FWPM_FIELD_IP_ADDRESS FWP_BYTE_ARRAY16_TYPE FWPM_CONDITION_IP_REMOTE_ADDRESS FWPM_FIELD_IP_ADDRESS FWP_BYTE_ARRAY16_TYPE FWPM_CONDITION_IP_LOCAL_ADDRESS_TYPE FWPM_FIELD_RAW_DATA FWP_UINT8 FWPM_CONDITION_IP_LOCAL_PORT FWPM_FIELD_RAW_DATA FWP_UINT16 FWPM_CONDITION_IP_REMOTE_PORT FWPM_FIELD_RAW_DATA FWP_UINT16 FWPM_CONDITION_IP_LOCAL_INTERFACE FWPM_FIELD_RAW_DATA FWP_UINT64 FWPM_CONDITION_INTERFACE_INDEX FWPM_FIELD_RAW_DATA FWP_UINT32 FWPM_CONDITION_SUB_INTERFACE_INDEX FWPM_FIELD_RAW_DATA FWP_UINT32 FWPM_CONDITION_FLAGS FWPM_FIELD_FLAGS FWP_UINT32 FWPM_CONDITION_INTERFACE_TYPE FWPM_FIELD_RAW_DATA FWP_UINT32 FWPM_CONDITION_TUNNEL_TYPE FWPM_FIELD_RAW_DATA FWP_UINT32 FWPM_CONDITION_CURRENT_PROFILE_ID FWPM_FIELD_RAW_DATA FWP_UINT32 FWPM_CONDITION_IPSEC_SECURITY_REALM_ID FWPM_FIELD_RAW_DATA FWP_BYTE_BLOB_TYPE FWPM_CONDITION_COMPARTMENT_ID FWPM_FIELD_RAW_DATA FWP_UINT32 FWPM_SUBLAYER_UNIVERSAL 14 FWPM_CALLOUT_IPSEC_INBOUND_TRANSPORT_V6 WFP Built-in IPsec Inbound Transport v6 Layer Callout Verifies that each received packet that is supposed to arrive over a transport mode security association arrives securely. FWPM_CALLOUT_FLAG_REGISTERED 00000260 FWPM_LAYER_INBOUND_TRANSPORT_V6 2 FWPM_CALLOUT_IPSEC_INBOUND_TUNNEL_V6 WFP Built-in IPsec Inbound Tunnel v6 Layer Callout Verifies that each received packet that is supposed to arrive over a tunnel mode security association arrives securely. FWPM_CALLOUT_FLAG_USES_PROVIDER_CONTEXT FWPM_CALLOUT_FLAG_REGISTERED 00000020 FWPM_LAYER_INBOUND_TRANSPORT_V6 6 {40916e9d-4803-4525-9ec7-889d8c1e9d11} IpV6 Ingress Transport Callout Rivet Filter Callout - IpV6 Ingress Transport Callout FWP_CALLOUT_FLAG_CONDITIONAL_ON_FLOW FWPM_CALLOUT_FLAG_REGISTERED {0ad5205a-0f51-4a94-8581-5e72a115b3aa} FWPM_LAYER_INBOUND_TRANSPORT_V6 289 {1625f947-add0-4039-8589-b0740a0c77d4} Rivet IpV6 Inbound Transport Filtering Layer  {0ad5205a-0f51-4a94-8581-5e72a115b3aa} FWPM_LAYER_INBOUND_TRANSPORT_V6 {6c03aa9b-e8f5-4545-9f0d-90c5397c2684} FWP_EMPTY FWP_ACTION_CALLOUT_TERMINATING {40916e9d-4803-4525-9ec7-889d8c1e9d11} 0 644623 FWP_UINT64 0 FWPM_LAYER_INBOUND_TRANSPORT_V6_DISCARD Inbound Transport v6 Discard Layer FWPM_LAYER_FLAG_KERNEL FWPM_LAYER_FLAG_BUILTIN FWPM_LAYER_FLAG_CLASSIFY_MOSTLY FWPM_CONDITION_IP_PROTOCOL FWPM_FIELD_RAW_DATA FWP_UINT8 FWPM_CONDITION_IP_LOCAL_ADDRESS FWPM_FIELD_IP_ADDRESS FWP_BYTE_ARRAY16_TYPE FWPM_CONDITION_IP_REMOTE_ADDRESS FWPM_FIELD_IP_ADDRESS FWP_BYTE_ARRAY16_TYPE FWPM_CONDITION_IP_LOCAL_ADDRESS_TYPE FWPM_FIELD_RAW_DATA FWP_UINT8 FWPM_CONDITION_IP_LOCAL_PORT FWPM_FIELD_RAW_DATA FWP_UINT16 FWPM_CONDITION_IP_REMOTE_PORT FWPM_FIELD_RAW_DATA FWP_UINT16 FWPM_CONDITION_IP_LOCAL_INTERFACE FWPM_FIELD_RAW_DATA FWP_UINT64 FWPM_CONDITION_INTERFACE_INDEX FWPM_FIELD_RAW_DATA FWP_UINT32 FWPM_CONDITION_SUB_INTERFACE_INDEX FWPM_FIELD_RAW_DATA FWP_UINT32 FWPM_CONDITION_FLAGS FWPM_FIELD_FLAGS FWP_UINT32 FWPM_CONDITION_INTERFACE_TYPE FWPM_FIELD_RAW_DATA FWP_UINT32 FWPM_CONDITION_TUNNEL_TYPE FWPM_FIELD_RAW_DATA FWP_UINT32 FWPM_CONDITION_CURRENT_PROFILE_ID FWPM_FIELD_RAW_DATA FWP_UINT32 FWPM_CONDITION_IPSEC_SECURITY_REALM_ID FWPM_FIELD_RAW_DATA FWP_BYTE_BLOB_TYPE FWPM_CONDITION_COMPARTMENT_ID FWPM_FIELD_RAW_DATA FWP_UINT32 FWPM_SUBLAYER_UNIVERSAL 15 FWPM_CALLOUT_WFP_TRANSPORT_LAYER_V6_SILENT_DROP WFP Built-in Silent Drop Transport v6 Discard Layer **Callout Implements stealth-mode filtering by silently dropping all incoming packets for which TCP does not have a listening endpoint.** FWP_CALLOUT_FLAG_ALLOW_OFFLOAD FWPM_CALLOUT_FLAG_REGISTERED 00000240 FWPM_LAYER_INBOUND_TRANSPORT_V6_DISCARD 18 {fc7cb985-9816-41ba-9cbd-e62c08edef19} Port **Scanning Prevention Filter This filter prevents port scanning. This many times means there are no listeners. If debugging ensure your scenario has one.** FWPM_FILTER_FLAG_HAS_PROVIDER_CONTEXT FWPM_FILTER_FLAG_HAS_FILTER_ORIGIN FWPM_PROVIDER_MPSSVC_WF 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

skintigh, 2025-01-16:
I used this command in powershell during a (failed) print. Nobody is listening on port 2021.

> Get-Process -Id (Get-NetTCPConnection -LocalPort 2021).OwningProcess

Get-NetTCPConnection : No MSFT_NetTCPConnection objects found with property 'LocalPort' equal to '2021'.  Verify the
value of the property and retry.
At line:1 char:18
+ Get-Process -Id (Get-NetTCPConnection -LocalPort 2021).OwningProcess
+                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (2021:UInt16) [Get-NetTCPConnection], CimJobException
    + FullyQualifiedErrorId : CmdletizationQuery_NotFound_LocalPort,Get-NetTCPConnection

Get-Process : Cannot bind argument to parameter 'Id' because it is null.
At line:1 char:17
+ Get-Process -Id (Get-NetTCPConnection -LocalPort 2021).OwningProcess
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidData: (:) [Get-Process], ParameterBindingValidationException
    + FullyQualifiedErrorId : ParameterArgumentValidationErrorNullNotAllowed,Microsoft.PowerShell.Commands.GetProcessC
   ommand

skintigh, 2025-01-16:
I did more testing and resmon.exe seems to show BS is listening on port 2021. Yet I also saw WFP drop packets like nobody was listening. ?!?

Worse, in resmon a failed print looks exactly the same as a print that was successful after pressing the "refresh" button a half dozen times. So I guess next step is to hunt down why you need to press refresh over and over, a problem that I think I read about in a post from 2023.

skintigh, 2025-01-16:
Yup, April 2023 https://forum.bambulab.com/t/lan-mode-issues-on-all-my-printers/9467

Good to know the bug driving me insane is at least 2 years old.

Lukrative525, 2025-01-20:
Can't say I understand everything going on, but I use my printer in LAN-only mode so I'm commenting just to help bring more attention to this.

TimNe0, 2025-02-14:
It would be good if this was fixed please.  We have 8 computers able to connect to 2 printers in a hackspace and it seems pot luck which will actually connect in LAN mode.

StoneLiBambu, 2025-03-14:
Thank you for your feedback. We will fix this issue in the next version.

Thanks!

Stone Li

skintigh, 2025-03-14:
That's great news! Is there a workaround for now?