bambulab/BambuStudio #5783: 斜拼接缝失效

LIA2333, 2025-01-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

A1 mini

### How to reproduce

取消勾选“智能斜拼接缝应用“， 打印圆柱零件。

### Actual results

在[Wiki](https://wiki.bambulab.com/zh/software/bambu-studio/Seam)里对智能斜拼接缝的描述是这样的：

_Bambu Studio具备智能选择接缝功能，根据模型的特点来选择是否应用斜拼接缝。当接缝位置的悬垂过大时，当接缝位置角度较小可以掩藏接缝时，也会选择不应用斜拼接缝。当墙壁没有合适的锐角以至于传统接缝无法有效隐藏的时候使用斜拼接缝。**关闭此选项则所有区域都应用斜拼接缝。**_

我取消了勾选，理论上应该所有区域都应用斜拼接缝，然而实际打印却没有斜拼接缝的效果:

[image]

同时，软件对斜拼接缝的预览也没有了！(这也证明了G代码里根本没有斜拼接缝)

[image]

### Expected results

修复斜拼接缝。

### Project file & Debug log uploads

[scarf seam falied.zip](https://github.com/user-attachments/files/18424540/scarf.seam.falied.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

QingZhangBambu, 2025-01-17:
您好，现在开关移动到了材料中，后期会给工艺加一个添加modifier的入口

QingZhangBambu, 2025-07-21:
请尝试新版本

closed 2025-07-21 by QingZhangBambu