bambulab/BambuStudio #5784: 阿基米德螺线作为顶层使用时，走线方向具有不确定性，有概率对产品的外观产生破坏

moe001, 2025-01-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

win10

### Additional system information

无关

### Printer

A1M/P1S

### How to reproduce

1. 载入附件工程
2. 进行切片

### Actual results

在预览界面拖动底侧的滑块查看顶层走线，可以看到各个方块的螺线，绘制方向不同，有的是由内到外，有的是由外到内。在由外到内的模型上，最终的打印质量不如由内而外的，因为热端在加工细小螺旋的过程中和完成后，会产生涂抹和刮擦，从而严重影响模型的光洁度。下面两张图可以看出绘制方向相反的两个方块。

（两种绘制方式似乎有一定的随机性，有时内向外多，有时外向内多，如果不能重现，可能需要多重启几次软件反复测试）

[image]

[image]

---
下面这张图是两个方向的模型打印的显微摄影，可以看到中央差别较大，其中由外到内的模型刮擦痕迹穿过了多条走线。

[image]

[image]

---
这张图是问题较为严重的一个宏观图，可以看到刮擦痕迹较大甚至产生了拉丝。

[image]

### Expected results

要么生成更好的走线路径规避涂抹（当前调整“Z抬升方式”等参数似乎效果不明显），要么统一由内到外画螺线。

另外可以额外提供一个额外的“双螺旋”填充样式，个人推测这样可以避免涂抹问题，但这样产生的色彩效果（尤其是我这种多色耗材）可能会不同，所以不能作为替代选项。

[image]

### Project file & Debug log uploads

[圆角矩形.zip](https://github.com/user-attachments/files/18425701/default.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

moe001, 2025-01-15:
另外，圆形之外的部分，现在的走线方向似乎也是随机的，这会影响多色耗材的色彩表现，希望能做到全都顺时针或者逆时针等更规则的走线方式。我不太确定是否有什么已有选项可以控制走线优化，但我似乎没找到。

[image]

XunZhangBambu, 2025-01-21:
@moe001 感谢您的反馈。针对您提出的剐蹭问题，可以通过关闭"减少填充回抽"功能解决。为了提高打印速度，我们默认开启了这个功能时，以减少在填充区域的回抽，因此不会触发Z向的抬升。对于走线的方向，我们以减少移动距离为目标，因此不同区域顺序可能不一致。我们在后续开发中会考虑您的建议，非常感谢！

[image]