bambulab/BambuStudio #5792: Problems with correct location of filament in slot on AMS Lite

Umar0784, 2025-01-16:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1 mini

### How to reproduce

1. When there is filament in all slots, Bambu Studio shows the correct slot number.
2. If I remove the filament in slot 1 and update AMS, Bambu Studio shows the wrong filament in the wrong slot. This only applies to the Bambu Studio program. On the printer says the correct location, same on the mobile app.
3. Bambu Studio shows the correct filament in the correct slot when I am in the "Device" tab, but when I go to "Prepare", this is where the error occurs.
4. See attached images

[image]

[image]

[image]

### Actual results

Problems with correct location of filament in slot on AMS Lite

### Expected results

Correct filament in the right slot, despite witch filament I add or remove

### Project file & Debug log uploads

[debug_network_Thu_Jan_16_00_44_31.log.zip](https://github.com/user-attachments/files/18446112/debug_network_Thu_Jan_16_00_44_31.log.zip)
[debug_network_Thu_Jan_16_00_49_00.log.zip](https://github.com/user-attachments/files/18446124/debug_network_Thu_Jan_16_00_49_00.log.zip)
[debug_network_Thu_Jan_16_00_57_50.log.zip](https://github.com/user-attachments/files/18446123/debug_network_Thu_Jan_16_00_57_50.log.zip)
[debug_network_Thu_Jan_16_15_52_19.log.zip](https://github.com/user-attachments/files/18446115/debug_network_Thu_Jan_16_15_52_19.log.zip)
[debug_network_Thu_Jan_16_18_23_42.log.zip](https://github.com/user-attachments/files/18446119/debug_network_Thu_Jan_16_18_23_42.log.zip)
[debug_network_Thu_Jan_16_18_25_11.log.zip](https://github.com/user-attachments/files/18446122/debug_network_Thu_Jan_16_18_25_11.log.zip)
[debug_network_Thu_Jan_16_18_26_18.log.zip](https://github.com/user-attachments/files/18446113/debug_network_Thu_Jan_16_18_26_18.log.zip)
[debug_network_Thu_Jan_16_18_31_04.log.zip](https://github.com/user-attachments/files/18446117/debug_network_Thu_Jan_16_18_31_04.log.zip)
[debug_network_Thu_Jan_16_20_02_55.log.zip](https://github.com/user-attachments/files/18446114/debug_network_Thu_Jan_16_20_02_55.log.zip)
[debug_network_Thu_Jan_16_20_11_59.log.zip](https://github.com/user-attachments/files/18446118/debug_network_Thu_Jan_16_20_11_59.log.zip)
[debug_network_Thu_Jan_16_20_49_52.log.zip](https://github.com/user-attachments/files/18446120/debug_network_Thu_Jan_16_20_49_52.log.zip)
[debug_network_Thu_Jan_16_21_01_17.log.zip](https://github.com/user-attachments/files/18446121/debug_network_Thu_Jan_16_21_01_17.log.zip)
[debug_network_Thu_Jan_16_23_18_02.log.zip](https://github.com/user-attachments/files/18446116/debug_network_Thu_Jan_16_23_18_02.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

bambu123, 2025-01-17:
Empty slot a skipped.

Umar0784, 2025-01-17:
> Empty slot a skipped.
> […](#)

Hi. Thanks for the feedback.
Can you explain a little more what you mean?

bambu123, 2025-01-19:
Empty slots are skipped.
The material list does not necessarily correspond exactly to AMS slots.

> From: ***@***.***>
> Date:  Sat, Jan 18, 2025, 05:14
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> > Empty slot a skipped.
> > …<#>
> Hi. Thanks for the feedback.
Can you explain a little more what you mean?

Umar0784, 2025-01-19:
> Empty slots are skipped.
> The material list does not necessarily correspond exactly to AMS slots.
>  From: ***@***.***>
>  Date:  Sat, Jan 18, 2025, 05:14
>  Subject:  Re: [bambulab/BambuStudio] Problems [...]

114 / 5 000
ok, thanks. How can I fix this problem?
As I said, it only applies in the program. Not in the printer display.

bambu123, 2025-01-20:
You don't have to fix the order in the prepare page.
When sending a task to printer, the project's filaments are mapped to real AMS slots.

> From: ***@***.***>
> Date:  Mon, Jan 20, 2025, 06:14
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> > Empty slots are skipped.
The material list does not necessarily correspond exactly to AMS slots.
From: @.>
Date:  Sat, Jan 18, 2025, 05:14
Subject:  Re: [bambulab/BambuStudio] Problems with correct location of filament in slot on AMS Lite (Issue #5792<https://github.com/bambulab/BambuStudio/issues/5792>)
To: @.>
Cc: @.>, @.>
> > > Empty slot a skipped.
…<#>
Hi. Thanks for the feedback.
Can you explain a little more what you mean?
> > > …<#>
> 114 / 5 000
ok, thanks. How can I fix this problem?
As I said, it only applies in the program. Not in the printer display.

Umar0784, 2025-01-20:
okay I understand what you mean but it becomes difficult to know which
filament is on which slot. Because if slot no. 1 has no filament, then I
have to change the location of the other filaments so that I can choose the
correct color from the other filaments. This got a bit complicated, but
could you please look at what I wrote about this issue in my post? I'm so
sorry :) What's strange is that I haven't had a problem with this before,
even if I don't use Bambu original filament. There is something not right.

bambu123, 2025-01-20:
There is also going to be a mapping when syncing AMS filaments.
That will keep colors in model.
Please wait.
> From: ***@***.***>
> Date:  Mon, Jan 20, 2025, 09:21
> Subject:  Re: [bambulab/BambuStudio] Problems with correct location of filament in slot on AMS Lite (Issue #5792)
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> 
okay I understand what you mean but it becomes difficult to know which

> 
filament is on which slot. Because if slot no. 1 has no filament, then I

> 
have to change the location of the other filaments so that I can choose the

> 
correct color from the other filaments. This got a bit complicated, but

> 
could you please look at what I wrote about this issue in my post? I'm so

> 
sorry :) What's strange is that I haven't had a problem with this before,

> 
even if I don't use Bambu original filament. There is something not right.

> 
>

Umar0784, 2025-01-20:
No sorry that hasn't happened since I got this problem. I will try to send
more pictures. What should I wait for? Just ask.