bambulab/BambuStudio #5807: crashes during slicing

bricha01, 2025-01-18:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

Device name	3D_Hub
Processor	13th Gen Intel(R) Core(TM) i7-13620H   2.40 GHz
Installed RAM	16.0 GB (15.7 GB usable)
Device ID	B7F86A7D-A2E0-4A34-AC7A-9E7215DD1528
Product ID	00342-22392-40903-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display
NVidia GeForce RTX 4060

### Printer

P1s

### How to reproduce

Load a detailed .stl file
Repair
Simplify
Slice

### Actual results

Once the slice gets to about 70 percent the app closes itself out. 

### Expected results

The slice should produce results and the app should not close.

### Project file & Debug log uploads

[bambu-studio.exe.412.zip](https://github.com/user-attachments/files/18461914/bambu-studio.exe.412.zip)

[gun_right.3mf.zip](https://github.com/user-attachments/files/18461929/gun_right.3mf.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

QingZhangBambu, 2025-01-24:
[image]
it could be sliced on version 1.10.2.64. could you try it?
and if it doesn't work. upload dmp files.

bricha01, 2025-01-24:
Thanks for getting back to me. The problem was I was slicing with a .2mm nozzle and the model created too many layers. I changed to a .4mm nozzle and it never happened again. Sent from my iPhone

image.png (view on web)

closed 2025-02-11 by QingZhangBambu