bambulab/BambuStudio #5809: STEP file import

simoneggersdorfer, 2025-01-18:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1C

### How to reproduce

1. Import a simple STEP file... nothing more

### Actual results

Whenever i try to slide or even input a different number for linear or angular deflection the system freezes and crashes. 
Its not a particularly big print at all. Whenever i open the dialogue it is set to the best settings, maybe thats the problem, but i cant change it since it always crashes.

### Expected results

Work normally? 

### Project file & Debug log uploads

Why?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Lukrative525, 2025-01-20:
Not a dev, but I would suggest you actually include the file you are trying to import so they can reproduce the issue. If you can't for privacy reasons, then model something else (that you don't care about) that also causes the crash and upload that.

For what it's worth, I also run windows 10, Bambu Studio version 1.10.1.50, and I just tried importing a reasonably complicated step file at the maximum precision (which, by the way is complete overkill). Even though it took a while, Bambu Studio was able to import it as a 22 million-triangle mesh without crashing:

[image]

MackBambu, 2025-01-22:
This issue has been present for a while, and most of the time it cannot be reproduced. In the next version, I will switch the multithreading solution from std::thread to boost::thread, which should improve stability.

closed 2025-03-31 by MackBambu