bambulab/BambuStudio #5812: Bambu should not lock down local device access

gonzoblue, 2025-01-18:
Bambu should not change the firmware to lock out local access from third parties. The spirit and community of 3D printing is about sharing and keeping things open and available to improve. If things get locked down people will stop using Bambu. I personally will remove all my models from Maker World and use Printables instead, and any new purchases will be Prusa machines. If you want to ruin your ecosystem then continue on with your new firmware changes. If you value what you have built, change the direction now and continue to allow third party direct access. There are secure ways to do this, you can't just blame "security" and make these changes.

vordenken, 2025-01-18:
Yes, this is a huge issue and I will tell everyone that BambuLab is not an option for a new 3D-printer anymore.
This is NOT acceptable!

Dexus, 2025-01-19:
I think everyone should open a issue with this topic so it will be seen from everyone!

Sangoku, 2025-01-19:
Jup. This is a HUGE red flag for me, Home Assistant won't work any more with it.  I will also stop posting on Maker world and will not buy their filament anymore. Have a p1s but if no open source.. il sell it and wont recomend it to anyone.

miikememe, 2025-01-20:
Broken "security" implementation forcefully coercing users into a workflow that will be monetized in the near future. Running ongoing cloud services isn't free, and they are playing the long game to lock us in and force us to subscribe.

Sad.