bambulab/BambuStudio #5813: mqtt command verification failed

corky008, 2025-01-18:
### Bambu Studio Version

1.08.03.00

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

did the beta update that you sent me now i get mqtt command verification error and now i can not print anything fthat i slice from my computor

### Printer

x1c

### How to reproduce

did updat tried slice project and will not send to printer now

### Actual results

does not work

### Expected results

does not work 

### Project file & Debug log uploads

fix problem

### Checklist of files to include

- [ ] Log file
- [ ] Project file

fahaaaxiki, 2025-01-19:
We found out that the Bambu Studio Version you submit is 01.08.03.00, you need to upgrade to [01.10.02.64](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.02.64) version, please re-check it and try again, thank you!

corky008, 2025-01-19:
Well that’s kind of hard when this is the only version it is allowing me to upgrade toSent from my iPhone

workman70, 2025-01-25:
I'm having the same issue.  The highest version I can download is 10.01.50.  I can't print anything now because of mqtt error

workman70, 2025-01-25:
I see now....downloaded from your link