bambulab/BambuStudio #5815: Unable to access live view via Studio application OR iOS phone application

pirate59, 2025-01-19:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 AND iOS

### Additional system information

I am not able to access the live view functionality within either the Bambu Studio application, or the iOS Bambu Handy application (v 2.17.1)

### Printer

P1P

### How to reproduce

Start any print
Click on live view

### Actual results

Click on live view icon, wait for 1 min, recieve a timeout error (0:00:10.00000 error) on IOS
When attempting on PC, the icon says initialising, but nothing happens.

### Expected results

click on live view, and see the stream.

### Project file & Debug log uploads

not relevant. happens on all prints

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2025-01-19:
Please update printer firmware. Old version has a bug.

pirate59, 2025-01-19:
Thanks for the quick response.

I want to note, that i have had the same firmware for a long time (approx 12 months), and this issue has occured in the last 2-3 weeks. 

While i realise the risks of old firmware and the updates, can you please share how a long running firmware suddenly stopped working with the live view?

Dexus, 2025-01-19:
> Thanks for the quick response.
>
> I want to note, that i have had the same firmware for a long time (approx 12 months), and this issue has occured in the last 2-3 weeks.
>
> While i realise the risks of [...]

Is not a bug in the firmware, the BUG name is BAMBULAB! Since they think they do things for "Security" and close their system for third parties and also your before free and open printer will now get forced to use there fucking cloud services. So I'm sure this is the bug @bambu123 means, not the firmware he wrongly mentioned.

bambu123, 2025-01-20:
Thank you for reporting. We'll check that.

miikememe, 2025-01-20:
"

> Please update printer firmware. Old version has a bug.
> […](#)

"
You broke your systems in order to force users to upgrade, which would lock out third-party open-source software, forcing us to use your "cloud" services when we have perfectly secure connections between devices in our LAN

bambu123, 2025-01-21:
No, you misunderstood. This is just a bug that exists in a specific version of the P1 firmware. Studio has abandoned compatibility with this bug since version 1.9.

> From: "Mike ***@***.***>
> Date:  Tue, Jan 21, 2025, 05:41
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> "
> > Please update printer firmware. Old version has a bug.
> > …<#>
> "
You broke your systems in order to force users to upgrade, which would lock out third-party open-source software, forcing us to use your "cloud" services when we have perfectly secure connections between devices in our LAN

Dexus, 2025-08-30:
Keep open