bambulab/BambuStudio #5821: can't move item up in bambu studio

Peakachu69, 2025-01-19:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11 pro, version 23H2, Build 22631.4751

### Additional system information

i just load any object i selected a preload square and just try to move it up and it would just bounce back down whats going on ? software has been restarted and computer tried different models and still same thing. 

### Printer

X1C

### How to reproduce

1. right click add primitive square
2. click on square 
3. click on move 
4. then move mouse over blue arrow and try to drag it up 

### Actual results

when dragging the blue arrow up when you let go of the mouse the image just drops back down to the original location you can move the object left and right just not up. 

### Expected results

it should move up when you drag it up . 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18470243/log.zip)

[Cube.zip](https://github.com/user-attachments/files/18470244/Cube.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-01-20:
Hello, our objects must land on the print plate. If you want one object to leave the print plate, you can combine it with another object and use the move gizmo to drag one of the objects along the Z-axis and then leave the print plate~

Peakachu69, 2025-01-20:
> Forwarded from ***@***.***
> Firefox Relay <https://relay.firefox.com/accounts/profile/>
> 0 email trackers removed
> Upgrade for more protection <https://relay.firefox.com/premium>
> [image: relay logo] <https://relay.firefox.com> Your dashboard

actually i can't if i combine the two objects i can only move one of them
as a single object and even then it won't let me move it up only down. when
i highlight all the items on the plate and right click it only shows
assemble once i do that i can only move everything as one item and it will
also not move up only down i mean why would u let it go down and not up ?
going down would also go pass the buildable plate. it all seems  so
confusing and hard to use why don't u just let it move were ever you want
and then have a button that just auto position them ? or have a option to
disable this feature.

Haidiye00, 2025-01-21:
[image]
Hello, holding down the alt button allows you to select a single combination of objects. Alternatively, you can select an object from the object list on the left.

[image]
Then click on the move gizmo ~