bambulab/BambuStudio #5823: Support Layers Printing Throughout Entire Model... New Bug or Am I Blind?!?

ScotRoss2, 2025-01-19:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro, Version 23H2, Build 22631.4751

### Additional system information

Processor	13th Gen Intel(R) Core(TM) i7-1360P   2.20 GHz
Installed RAM	16.0 GB (15.5 GB usable)
System type	64-bit operating system, x64-based processor

### Printer

Bambu Labs P1S

### How to reproduce

1. Designed - in Bambu Labs Studio - a basic cube with a overhang that requires support.
2. Used many different print profiles (all reproduce this problem), but for this purpose, I used the system presets - any of them.
3. Tried on 3 different P1S printers, same result.
4. When the model slices, there is only support material (Bambu Labs PLA Support Material that came with the printer) showing where the support is required - under the overhang.
5. Tried many different Support settings, even the default settings.
6. Tried many previously successful prints that utilized support material and they all fail with support material layers being printed throughout the entire model.

### Actual results

When printed, the support material layers are printed through the entire model. See attached images.

### Expected results

Support material printed only under the overhangs, as indicated by the preview, and my settings.

### Project file & Debug log uploads

[SupportLayerTest.zip](https://github.com/user-attachments/files/18470495/SupportLayerTest.zip)

[Pics.zip](https://github.com/user-attachments/files/18470563/Pics.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

NothFie, 2025-01-22:
@ScotRoss2 Thanks for your report. We tested this issue with many printers and only one printer reproduced it. Then we found that that printer did not have enough flushing during filament replacement, and the residual amount of the previous filament was printed on the model, causing this problem. In this situation, I suggest you try cleaning the carbon rods(which may have caused a decrease in accuracy) or increasing the amount of flushing by change prime volume. If you still encounter problems after trying, please provide a Gcode file and we will further analyze the cause of the problem for you。

ScotRoss2, 2025-01-26:
Hi, I've tried the flush setting and all 3 printters are practically brand new. I then added a long primitive to see if the support layers "fade to black" to test the residual filament in pressure chamber and nozzle theory (all 3 printers at the exact same time doesn't support this very well, but it's slightly plausible.)

Atta he'd is a photo of the new test piece  showing that the support filament is definitely being printed.

I will get the gcode from all of my machines soon.

Thanks!

[image]

ScotRoss2, 2025-01-26:
Here's the gcode for this model.

I used Bambu Studio for one print and Orca Slicer for the other - same results - it's gotta be something I did while poking around in the advanced settings for the nozzle profile. Does changing a setting in there affect ALL printers? It seems to me it does, since when I switch printers in the Studio, the nozzle preset is still set to the same profile.

[Support Filament Issue.zip](https://github.com/user-attachments/files/18548547/Support.Filament.Issue.zip)

ScotRoss2, 2025-01-26:
I've scoured the gcode and it does appear that the "Support Interface" layers are printing correctly. So I changed the "Flushing Volumes" from the default of "0.00" to "0.50" and hit "Recalculate". I then printed on one of my machines and it WORKED! Well, for the most part. You can still see that the lines are different, but they did print in black filament, just slightly "off", like the nozzle was bumped by a huma hair's thickness, but ONLY for the corresponding support interface layers - the rest of the print is perfect. So 100% back to what it was prior to this issue, nope, but I'll take 99%!

The only thing that baffles me, is "why?" (times 3!)
Why have I been printing fine without this setting and suddenly - BAM - now I need this setting?!? Then:
Why on 3 P1S machines and one A1 machine? And finally:
Why all machines at the _exact same time_ suddenly need this setting?

I can possibly see the 3 P1S machines being affected by some strange coincidence, but throw in a 4th, completely different model of machine (the A1) that I've owned and have been printing with for almost a year and these P1S machines are literally only a month old with maybe 30 hours of print time on average.... pretty strange coincidence, and I don't believe in coincidences. _Something_ caused this and I will keep digging to find out why this is happening. 

My next tests will include:
1. Resetting the Flushing Volumes back to 0.00
2. Replace all hotends with new
3. Set up print profiles with different flushing settings
4. I also moved the Prime Tower far away from the model to see if there was any change, so I'll reposition the tower near the model as well, just to see if there's a difference. 
5. Maybe resize the Priming Tower as well? Scale it up a bit?

I'll post my findings here, but I think I can at least print my non-production parts for now. I do not like the look of the slightly offset lines on these Support Interface layers (where there's no support filament, but on the same layer as the support interface), as it would be noticeable to my customers. I still have some tweaking to do get rid of that before I can get back into production.

Thanks

closed 2025-02-24 by NothFie