bambulab/BambuStudio #5827: Bambu Slicer crashing when slicing complicated models

SignorLuigi, 2025-01-20:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro - 24H2 - OS build 26100.2894

### Additional system information

DELL, XPS 8940
11th Gen Intel(R) Core(TM) i9-11900K @ 3.50GHz   3.40 GHz
64 GB Ram
64-bit operating system, x64-based processor

I am trying to print [this LTS](https://makerworld.com/en/models/665413?from=search), but when I click in "Previews" or "Slice Plate" the BambuLab crashes without saying anything, it just close the program.

I have tried to Set Affinity and it didn't work, using saved mode and it didn't work, run it from CMD with some extra command and it didn't work, all those potential temporary solutions are [from here.](https://forum.bambulab.com/t/bambu-slicer-orca-slicer-crashing-when-slicing-complicated-models/61424/60)

 My only solution was to scale down from 100% to 99% and that worked.

Please let me know if I can help more.
Thanks

### Printer

Bambu LAB PS1 

### How to reproduce

1) Get STL from [here](https://makerworld.com/en/models/665413?from=search)
2) Open with Bambulab
3) Click Preview, or Slice Plate at 100% scale or more.
4) If it crashes then scale down to 99% or less.
5) Then that will work

### Actual results

I explained in the step above, the software crashes without any warning

### Expected results

I want to be able to print at ANY size....

### Project file & Debug log uploads

https://makerworld.com/en/models/665413?from=search

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2025-01-24:
[image]
i use 1.10.2.64, it could be sliced. Could you try it?
and if it does not work, upload 3mf and dump files. I will check it.

SignorLuigi, 2025-01-24:
Hi, I tested that version and still has the same problem: the app crashes and closes it without any warning when clicking in "Preview" or "Slice Plate". (both has the same effect)

""I am attaching the Zip file with all the files of the 3d model, Open it as it is, do not modify it, and then click on preview.""

Also I am attaching the crash files/dumps that I believe are in the "AppData\Roaming\BambuStudioBeta\log".

Please, let me know if you need something else.
Thanks

[Mate+F1,+Polimero+Escalonado.zip](https://github.com/user-attachments/files/18535654/Mate%2BF1.%2BPolimero%2BEscalonado.zip)

QingZhangBambu, 2025-01-26:
@SignorLuigi uploading a 3mf file would be helpful. the crash may only occur on certain parameters

SignorLuigi, 2025-01-27:
[F1.zip](https://github.com/user-attachments/files/18552167/F1.zip)

Here is the F1.3mf that I created using the stl one. Sorry I thought it was on the first zip file.
Thanks

QingZhangBambu, 2025-01-27:
it crashed on infill generator.  i will asked concerned staff to check this.

[image]

QingZhangBambu, 2025-02-07:
@XunZhangBambu please check this.