bambulab/BambuStudio #5831: Crashing issue

Grubbin615, 2025-01-21:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11 home

### Additional system information

CPU: AMD Ryzen 7 6800H with Radeon Graphics 3.20 GHz
RAM: 16 GB
GPU: NVIDIA GeForce RTX 3060 Laptop GPU

### Printer

Bambu X1 Carbon

### How to reproduce

1. Open app.
2. Load online model from MakerWorld
3. Adjust settings of the print.
4. Crash

### Actual results

It closes out of the app.

### Expected results

It should not crash

### Project file & Debug log uploads

not doing that, asking way too much.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-01-21:
Thank you for your feedback. Could you please provide the log and DMP files? We'll analyze them here.

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

Haidiye00, 2025-04-15:
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50
please try it

closed 2025-04-15 by Haidiye00