bambulab/BambuStudio #5836: Painted Brims do not stay if objects are merged to use variable layer height

mhwunder, 2025-01-21:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS 13.7.2

### Printer

P1P open

### How to reproduce

4 objects on plate.  Not tall objects but objects with smaller contact surface area.

Paint brims for two of the objects.

Merge objects to use variable layer height.

Slice print

### Actual results

Painted brims are not printed/used.

[image]

### Expected results

Painted brims would still print.

### Project file & Debug log uploads

[No painted brim.3mf.zip](https://github.com/user-attachments/files/18495215/No.painted.brim.3mf.zip)

No log file for current date.

### Checklist of files to include

- [ ] Log file
- [x] Project file

MackBambu, 2025-01-22:
1.Paint brim ? support?
2.Why merge objects?
It would be best to record an operation video to reproduce the steps.

MackBambu, 2025-01-24:
Yes, when merging the current objects, the brim ears are removed because the positional mapping gets messed up. This issue will be fixed later.
Is it possible to merge the model first and then draw the brim ears afterward?

mhwunder, 2025-01-24:
Doing after merge did work if I manually paint them in.  Obviously, if I use the auto generate, it put brims on all objects in the merge.  

I did note something else.  I had a lack of support filament used for the support/raft interface on the two smaller tires after variable layer height adjustment.  Dual tires after painting in support had good use of the support filament for the raft interface.

No Variable Layer Height adjustment
[image]

After Variable Layer Height adjustment (Used adaptive Quality/Speed set at .20)
[image]

MackBambu, 2025-01-24:
The principles for support painting and brim ears painting differ significantly. Brim ears painting is more akin to adding connectors in a Boolean cut tool, so it's difficult to make them behave exactly the same way.