bambulab/BambuStudio #5841: Text rendering gets cut off in certain calibration page locations

Nightfirecat, 2025-01-23:
### Bambu Studio Version

1.10.1 Public Release (flatpak)

### Where is the application from?

Built from source by yourself or others

### OS version

Linux - Debian 12

### Additional system information

hardware n/a
Desktop environment is KDE Plasma 5.27.5 on X11

### Printer

Bambu Lab A1 Mini

### How to reproduce

1. Go to Calibration tab
2. Observe text is cut off in several places, see results below

### Actual results

[image]

### Expected results

Circled areas should not have text cut off

### Project file & Debug log uploads

[debug_Wed_Jan_22_23_31_54_2.log](https://github.com/user-attachments/files/18516965/debug_Wed_Jan_22_23_31_54_2.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Nightfirecat, 2025-01-23:
Of note, this was originally reported at https://github.com/flathub/com.bambulab.BambuStudio/issues/29, where I also found a similar text clipping issue in the "Manage Result" dialog.

zhimin-zeng-bambulab, 2025-02-05:
Thanks for reporting this issue, we will investigate and try to reproduce this issue.

Nightfirecat, 2025-08-31:
This issue still exists as described, now on Debian 13 running KDE 6.3.6 on X11.