bambulab/BambuStudio #5845: Turn, Scale and move Anchors gets VERY small

KimA998, 2025-01-23:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

Quite often the anchors when moving, scaling or turning an object get VERY small and hard to hit with the mouse.

Only solution is to close Bambu Studio and restart it.

### Printer

X1C

### How to reproduce

I haven´t figured it out yet how it happens - but it does!

### Actual results

You need to restart Studio to have "normal" size anchors again.

### Expected results

I didn´t expect the anchors to change in size!!

### Project file & Debug log uploads

[Nimbus Print 1.1.zip](https://github.com/user-attachments/files/18523367/Nimbus.Print.1.1.zip)

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-01-24:
Hello, we will try to reproduce it. If you can find a stable way to reproduce it, you can record a GIF and send it to us. Thank you very much.

And can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-03-26 by Haidiye00