bambulab/BambuStudio #5848: Bambu Studio -> Device -> AMS view not updated properly after filament change

rmilkowski, 2025-01-24:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 18.3

### Additional system information

MacOS 18.3 M1

### Printer

X1C

### How to reproduce

When Device -> Status tab is activated during printing, after a filament change the AMS view on the right does not fully update the line showing which slot/filament is in use. Most of the line is kept with the color of previous filament (yellow), and just a small part next to the extruder (and its circle) updates with the new color (black). If I click on Device->MicroSD Card and then go back to Device->Status then the AMS view is correctly draped with the entire line showing the new filament. See the two attached screenshots - one with most of the Eline showing the previous (yellow) filament and one after going to different view and back where the new filament is shown correctly (black).

Also the PA-CF text in AMS/Slot4 is shown in two lines - the font should be adjusted so it is showed on a single line like other filaments.

Same issue was observed on the latest stable release, not just on beta.

### Actual results

AMS diagram not updated correctly after filament change - most of the line is showing the previously used filament.

Also the PA-CF text in AMS/Slot4 is shown in two lines - the font should be adjusted so it is showed on a single line like other filaments.

### Expected results

Without having to change to another view and then going back to the Status view, the AMS diagram should be fully updated after a filament change.

The PA-CF text in AMS/Slot4 shouldn't wrap to other line and be displayed constantly with other slots/filaments.

### Project file & Debug log uploads

[image]

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XinZhangBBL, 2025-04-10:
@rmilkowski  Sorry for the issue, please try using the latest version
https://github.com/bambulab/BambuStudio/releases/download/v02.00.01.50/Bambu_Studio_win_public-v02.00.01.50-20250328181949.exe

XinZhangBBL, 2025-04-25:
Halo, The problem solved or not ? If there is no reply, we will close this issue。

rmilkowski, 2025-04-25:
Yes, the latest Bambu 2.x has the issue fixed. Thank you!

closed 2025-04-25 by rmilkowski