bambulab/BambuStudio #5849: Studio will not display model on sd card. (Same issue on the Android App)

mitcheip, 2025-01-24:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

Bambu P1S

### How to reproduce

go to device
click on "Micro SD Card"
Time lapse shows "File does not exist"[10]
Click on "Model"
"File does not exist"[10] is displayed

### Actual results

[image]

### Expected results

I'm expecting to see the print files on the printer micro sd card, (and the time-lapse from print)

### Project file & Debug log uploads

[debug_Fri_Jan_24_15_30_14_27932.log.zip](https://github.com/user-attachments/files/18538906/debug_Fri_Jan_24_15_30_14_27932.log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gaolegao-lx, 2025-03-24:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Phil-Barker, 2025-04-16:
Likely related to https://github.com/bambulab/BambuStudio/issues/6235